h2zero/esp-nimble-cpp #161: How to get connection ID from a NimBLEConnInfo?

cmorganBE, 2024-05-20:
There is NimBLEServer::disconnect(uint16_t connID, uint8_t reason), but I don't see a way to get the connection is from the given NimBLEConnInfo. Is this the handle returned from getConnHandle()?

I might suggest we make a type for this value then, like ClientID, and then use that type also in NimBLESErver::disconnect() (and maybe add an overload for disconnect() that takes a NimBLEConnInfo reference. I can open a PR if one or both of these make sense and the first paragraph is correct.

I'd like to force disconnect the device in onConnect() if it isn't in the whitelist (as a work around for the RPA issue I'll let all clients connect and then disconnect them if they aren't in the whitelist after getting the public and RPA addresses to check), but not sure how to get the connection ID.

cmorganBE, 2024-05-21:
It looks like the getConnHandle() value is the connection ID. @h2zero would you be ok with a PR to make that value a type so the types can be aligned, and add a helper function or switch the disconnect() to take a NimBLEConnInfo&?

h2zero, 2024-05-22:
@cmorganBE The connection ID can be retrieved from the server callbacks or via the functions defined here: https://github.com/h2zero/esp-nimble-cpp/blob/226c67f729b049498e95eb68cc97673f4a214de2/src/NimBLEServer.h#L77

cmorganBE, 2024-05-22:
@h2zero connection id isn't the same as NimBLEConnInfo->getConnHandle()?

cmorganBE, 2024-05-23:
@h2zero opened #162 that would be what would have helped me and should help the next person with the same question.

cmorganBE, 2024-05-28:
@h2zero did that look ok?

closed 2024-06-08 by h2zero