h2zero/esp-nimble-cpp #163: Issue with Switching Services

michaelboeding, 2024-05-25:
I'm running into an issue where I start provisioning my device using esps prov api, which uses its own services. When I go out of that later in the application I create my services, but it seems like IOS is caching these values until I reset my network settings on the device. Which isn't a great solution for customers, is there anyway to make the client not read from  the cache on a new connection with the server? Or some other solution to this?

h2zero, 2024-06-04:
The basic required services are automatically created by the stack, no need to define them yourself.

michaelboeding, 2024-06-04:
I never see those services when I run a BLE sniffer though? Should they automatically appear as services?

h2zero, 2024-06-04:
They show in nrfConnect on android but not iOS. iOS automatically subscribes to the service changed characteristic.

michaelboeding, 2024-06-04:
Got it - the weird thing is I never get the callbacks for services changed...has it ever been tested? 

'''swift

    // This method is called when the peripheral's services are modified
    func peripheral(_ peripheral: CBPeripheral, didModifyServices invalidatedServices: [CBService]) {
        print("Services modified: \(invalidatedServices)")
        
        // Handle the modified services
        for service in invalidatedServices {
            print("Service \(service.uuid) was invalidated")
            
            // Rediscover the services if necessary
            peripheral.discoverServices([service.uuid])
        }
    }

'''

h2zero, 2024-06-04:
That is all handled by the stack at a lower level, not presented to the app code. With full debug logging turned on you can see iOS subscribe to it. If you make a change to the services the indication will be sent by the stack as well.

michaelboeding, 2024-06-04:
I'll try to run it in full debug - I'm never receiving it currently in CoreBluetooths callbacks. I'm also running the latest 5.2.1 of the IDF but I can't confirm that it worked previously either.