h2zero/esp-nimble-cpp #173: No host/ble_gap.h file when building with platformio / arduino

NorbNorb, 2024-07-02:
My ESP32 project built with platformio used the NimBLE-Arduino library but I finally went into RAM issues. So I hoped to find a more lightweight solution here.

Using VS Code and platformio, I try to switch from **NimBLE-Arduino** to **esp-nimble-cpp**. I therefore simply changed the _lib_deps_ in the _platformio.ini_ file (see below).

However, I get this error:
**.pio/libdeps/dfrobot_firebeetle2_esp32e_16MB/esp-nimble-cpp/src/NimBLEUtils.h:16:10: fatal error: host/ble_gap.h: No such file or directory**

I read [here](https://github.com/h2zero/esp-nimble-cpp/issues/60#issuecomment-1722768106) that it might be resolved by changing the platform from _arduino_ to _espidf_ but this lead me down the rabbit hole, leading to further errors.

Here's my full **platformio.ini** file:

```
[platformio]
default_envs = dfrobot_firebeetle2_esp32e_16MB

[env]
framework = arduino
build_flags = '-Wall'
lib_deps =
  adafruit/Adafruit BusIO @ ^1.15.0
  adafruit/Adafruit Unified Sensor @ ^1.1.14
  bblanchon/ArduinoJson @ ^7.1.0
  zinggjm/GxEPD2 @ ^1.5.7
  ;h2zero/NimBLE-Arduino@^1.4.2
  https://github.com/h2zero/esp-nimble-cpp.git#master

[env:dfrobot_firebeetle2_esp32e_16MB]
platform = espressif32
board = dfrobot_firebeetle2_esp32e_16MB
monitor_speed = 115200
board_build.partitions = default_16MB.csv
debug_tool = esp-prog
```

I have spent a whole day on this but couldn't resolve it.
Do you have any tips on how to make it work?

h2zero, 2024-07-02:
Why are you switching from NimBLE-Arduino? If you are using arduino framework you should not use this repo unless you are using arduino as a component as well `framework = espidf, arduino`.

NorbNorb, 2024-07-03:
I am trying to switch from NimBLE-Arduino to this repo to see if it uses less space in RAM. My free heap when I include the arduino library is too small for my main tasks. So I hope that by switching I will have more memory available. Please tell me if this is not the case.

h2zero, 2024-07-03:
This repo and NimBLE-Arduino are the same code, the only difference being that NimBLE-Arduino includes the entire BLE stack code. 

Switching to this repo may help, but only if you switch to espidf entirely with your project, which opens up configuration options.

NorbNorb, 2024-07-03:
I'll give it a try to switch from arduino to espidf.
Here's BTW what the free heap space looks like without 
[image]

and with my (simple) BLE code:
[image]

closed 2024-11-20 by h2zero