h2zero/esp-nimble-cpp #174: whitelisted connections do not persist across power cycles

cmorganBE, 2024-07-02:
Using https://github.com/cmorganBE/esp-nimble-cpp-test/tree/bonding_with_whitelist_control_and_sc (latest esp-nimble-cpp), esp-idf v5.2.2

BT_NIMBLE_NVS_PERSIST is y

Program with:

```
idf.py build erase-flash flash monitor
```

Steps to reproduce:
1. Connect with lightblue iOS during 'Any'
1. Disconnect lightblue and confirm it can connect during 'Whitelist only' mode
1. Power cycle esp32s3-devkit-c1
1. Lightblue unable to connect during 'Whitelist only' mode.

@h2zero I thought this was working before but maybe I forgot or didn't test it? Same arrangement as before works in terms of hourly, if that helps expedite looking into this.

h2zero, 2024-07-02:
@cmorganBE The whitelist is not persisted in NVS by the stack as far as I know. You would need to populate it on startup with your own persisted list or just get all the addresses of the bonded devices.

cmorganBE, 2024-07-02:
So the bonding is persisted, as indicated by the menuconfig option but not whitelists? Weird….

CAUTION: This email originated from outside the organization. Do not click links or open attachments you were not expecting.

h2zero, 2024-07-02:
Yes, that's how I perceive it, I think it's because whitelists change more frequently.

cmorganBE, 2024-07-02:
Ahh k. I can’t find any info on it being persisted so I think it isn’t. I’ll implement persisting it here and see how it goes.

cmorganBE, 2024-07-08:
@h2zero for serializing and deserializing the whitelist addresses, I presume the NimBLEAddress type has to be preserved and restored as well? I'm going to start down that path, assuming its necessary, until I hear otherwise.

h2zero, 2024-07-08:
Yes the address type is necessary. I would just store the NimBLEAddress used as a blob and read it back on restart.

cmorganBE, 2024-07-08:
@h2zero I thought of going that route but how to avoid having too many whitelisted devices? Not that anyone might pair that many but thought it made sense to have some upper limit. And then if there is a limit, how to ensure the new ones replace the old ones? I was going to store an array of N entries in a blob but yeah NimBLEAddress may be POD and safe to persist...

h2zero, 2024-07-08:
The whitelist has a limit of 12 I believe. I would erase whitelist entries by comparing with the bonded device list and delete any addresses that are no longer bonded.

cmorganBE, 2024-07-08:
@h2zero is the assumption that there is a limit to bonded devices as well? If so that approach should work. Store in a single blob or separate nvs entries (then they have to have unique names which could be the address but that seems wasteful...)

h2zero, 2024-07-08:
Yes there is a bond limit, defaults to 3, you can change it in the config.

cmorganBE, 2024-07-08:
@h2zero safe to assume the whitelist addresses == bonded addresses?

finger563, 2024-07-08:
I use bonded addresses as my whitelist addresses and it works as intended. No additional data needs to be stored.

cmorganBE, 2024-07-08:
@finger563 doh, that's an even better idea.... wow thank you for that

cmorganBE, 2024-07-08:
@finger563 and that works since the bonding stores the address type? I'm unsure of how the type is used so it wasn't clear that the bonded address would know its type like the whitelist might require.

finger563, 2024-07-08:
No address type is required, I just load the bonded address and create a NimBLE address from it and pass that to the whitelist

cmorganBE, 2024-07-08:
let me whip that up and give it a shot

finger563, 2024-07-08:
```c++
            auto paired_addresses = ble_gatt_server_.get_paired_devices();
            for (const auto &address : paired_addresses) {
                logger_.info("Adding {} to whitelist", address.toString());
                NimBLEDevice::whiteListAdd(address);
            }
```
Where my server implements this:
```c++
  /// Get the paired devices
  /// @return The paired devices as a vector of Addresses.
  std::vector<NimBLEAddress> get_paired_devices() {
    std::vector<NimBLEAddress> paired_devices;
    auto num_bonds = NimBLEDevice::getNumBonds();
    for (int i = 0; i < num_bonds; i++) {
      auto bond_addr = NimBLEDevice::getBondedAddress(i);
      paired_devices.push_back(bond_addr);
    }
    return paired_devices;
  }
```

cmorganBE, 2024-07-08:
That's very similar to what I have here:

```
        for(int i = 0; i < NimBLEDevice::getNumBonds(); i++)
        {
            auto address = NimBLEDevice::getBondedAddress(i);
            NimBLEDevice::whiteListAdd(address);
        }
```

cmorganBE, 2024-07-08:
yep, that appears to be working well here, thank you much @finger563

h2zero, 2024-07-09:
Yes, that is by far the best way to go about this.

cmorganBE, 2024-07-10:
Closing as fixed!

closed 2024-07-10 by cmorganBE