h2zero/esp-nimble-cpp #189: deleteClient() not returning

xieliwei, 2024-07-29:
I'm working with an ESP32-S3 on IDF 4.4.7 using platform IO with Arduino as a component.

My use case requires regularly connecting to a bunch of battery devices and reading the battery level characteristic. The code looks like this:
```
class BLEScannerClientCallbacks : public NimBLEClientCallbacks {
    void onConnect(NimBLEClient *client) {}
    void onDisconnect(NimBLEClient* pclient) {}
};

bool BluetoothBatteryCheck(BLEAddress device, uint8_t *battery_level) {
  NimBLEClient *client = NimBLEDevice::createClient();
  client->setClientCallbacks(new BLEScannerClientCallbacks());
  if (!client->connect(device)) {
    NimBLEDevice::deleteClient(client);
    return false;
  }

  NimBLERemoteService *service = client->getService(BLEBatteryService);
  if (!service) {
    NimBLEDevice::deleteClient(client);
    return false;
  }

  NimBLERemoteCharacteristic *characteristic = service->getCharacteristic(BLEBatteryCharacteristics);
  if (!characteristic) {
    NimBLEDevice::deleteClient(client);
    return false;
  }

  if (characteristic->canRead() && battery_level) {
    *battery_level = characteristic->readValue<uint8_t>();
  }

  NimBLEDevice::deleteClient(client);
  return true;
}
```

What I find is that sometimes deleteClient() does not return. I think usually because the device has moved out of range or a stray device not intended to be interrogated got picked up (so getting the service fails).

I've tried a few mitigation methods modifying deleteClient() in NimBLEDevice.cpp:
```
while(pClient->isConnected()) {
    pClient->disconnect(); // Always try disconnecting
    taskYIELD();
}
```
but that did not help.

If I force the endless while loop with a maximum number of tries:
```
uint16_t tries = 65535;
while(pClient->isConnected() && tries--) {
    pClient->disconnect();
    taskYIELD();
}
```
We of course stop getting stuck, but it seems like the affected device never gets picked up again (no scan callback triggered for it, but device is definitely still operating). Other devices continue to be processed though, so at least my application does not 'hang' in this case.

The scan is activated this way:
```
if (!NimBLEDevice::getInitialized())
  NimBLEDevice::init("");
_scan = NimBLEDevice::getScan();
_scan->setScanCallbacks(new BLEScannerAdvertisedDeviceCallbacks(), true);
_scan->setFilterPolicy(BLE_HCI_SCAN_FILT_NO_WL);
_scan->setMaxResults(0);
_scan->setActiveScan(true);
_scan->setInterval(1200);
_scan->setWindow(1200);
while (1) {
  _scan->start(10 * 1000, false);
  vTaskDelay(pdMS_TO_TICKS(11 * 1000));
  _scan->stop();
  _scan->clearResults();
}
```

I don't see anyone else having this problem, so I think I must be setting something up wrong. Can anyone point me in the right direction?

xieliwei, 2024-07-29:
Oh, I should mention that although it seems that having a failed link (device out of range, stray device, service/characteristic read fail). is a pre-requisite for this happening, it does not imply that every failed link leads to a hang.

In other words, I regularly have devices moving out of range, etc, but only rarely does it cause deleteClient() to not return; It sometimes occurs shortly (seconds) after boot, but it also happens hours or even days later.

h2zero, 2024-07-29:
This reminds me of an old problem where the ble controller would not disconnect or at least not send the event. A timer was added to reset the host after 2 seconds if the disconnect event didn't occur. I thought the controller was fixed by now though, what IDF version are you using?

xieliwei, 2024-07-29:
4.4.7, which is the latest pre-5.0 IDF supported by Arduino. I think there is a 4.4.8 now, which I can upgrade to, but I don't think that makes a difference.

I saw the timer implementation, but can't really figure out how the timer, `npl_freertos_callout_reset()` and `ble_gap_terminate()` work. There's a lot of chatter to filter through at the debug log level, so I need some time to figure out what happens with a failing disconnect().

Regarding using new hardware, that is certainly odd. I'm seeing this on all units, and we order the modules from Mouser, so I doubt they're of dubious quality. But I'll give it a try with a devkitC instead and report back.

h2zero, 2024-07-29:
`npl_freertos_callout_reset()` starts the timer, after it expires it calls this:
https://github.com/h2zero/esp-nimble-cpp/blob/209f70a0839b18e59e97517c7c9276eefe53430b/src/NimBLEClient.cpp#L112
`ble_gap_terminate()` tells the nimble host to terminate the connection. 
This problem is difficult because the error stems from the controller not disconnecting and the source for that is closed.

xieliwei, 2024-07-29:
Okay, I think I see one possible cause. Here's the log sequence when this scenario occurs:
```
BluetoothBatteryCheck(): [BLE] create client
BluetoothBatteryCheck(): [BLE] connect device xx:xx:xx:xx:xx:xx
NimBLEClient: Connected event
[Something happens here]
NimBLEClient: Connect timeout - no response
NimBLEClient: Connection ID not found
BluetoothBatteryCheck(): [BLE] couldn't connect to device xx:xx:xx:xx:xx:xx to read battery level
[My code tries calling deleteClient() here]
NimBLEClient: Connection ID not found
[Hangs]
```

In `[Something happens here]`, something happens due to the marginal connection (and presumably failed MTU negotiation) that removes the connection from the internal connection table.

Just to be sure, I added extra logging to the BLE_GAP_EVENT_DISCONNECT case in handleGapEvent() (since for performance reasons I am only able to log up to `info` but not `debug`)
https://github.com/h2zero/esp-nimble-cpp/blob/209f70a0839b18e59e97517c7c9276eefe53430b/src/NimBLEClient.cpp#L994

and can confirm that BLE_GAP_EVENT_DISCONNECT is never called.

And to be very sure, I logged the connection ID when `Connection ID not found` occurs. It is not `BLE_HS_CONN_HANDLE_NONE` but an actual connection ID.

Thus isConnected() will return true, but disconnect() does not see a valid connection ID, and does not set up the timer. Instead, it returns BLE_HS_EALREADY.

deleteClient() will receive BLE_HS_EALREADY and thus proceeds to wait for isConnected() to change, but nothing will happen, and we get stuck.

Consequently, even if we force our way out of the stuck loop, the device will never show up again since BLE_GAP_EVENT_DISCONNECT does not fire, thus the device is not removed from the ignore list added during connect:
https://github.com/h2zero/esp-nimble-cpp/blob/209f70a0839b18e59e97517c7c9276eefe53430b/src/NimBLEClient.cpp#L1018

I can currently workaround these by limiting the number of loop iterations and calling removeIgnored() after the loop, but I'm not sure if what I'm experiencing is due to a change in IDF behaviour or a bug in my code.

xieliwei, 2024-07-30:
I realised I should look into the events in more detail. Turns out BLE_GAP_EVENT_DISCONNECT is being emitted, but very strangely:
```
NimBLEClient: Got Client event BLE_GAP_EVENT_CONNECT 
NimBLEClient: Connected event
NimBLEUtils: gapEventToString: Unknown event type 25 0x19
NimBLEClient: Got Client event Unknown event type
NimBLEClient: Got Client event BLE_GAP_EVENT_CONNECT 
NimBLEClient: Got Client event BLE_GAP_EVENT_DISCONNECT
NimBLEClient: rc = 534, m_conn_id = 0x49ff, conn_handle = 0x0001
NimBLEClient: Connect timeout - no response
NimBLEClient: Connection ID 0x0001 not found
```

The rc, m_conn_id and conn_handle values lead to this return when BLE_GAP_EVENT_DISCONNECT is emitted:
https://github.com/h2zero/esp-nimble-cpp/blob/209f70a0839b18e59e97517c7c9276eefe53430b/src/NimBLEClient.cpp#L998-L1012

xieliwei, 2024-07-30:
Aha! I think I have this solved. The invalid `m_conn_id` points to some form of corruption, and event type 25 points to a `BLE_GAP_EVENT_REATTEMPT_COUNT` event.

So the first and main reason why I'm facing this issue is because I have nimBLE configured (`CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT`) to do connection re-attempts on failure. This causes two things to occur:
1. On re-attempt, the old connection is automatically closed and a new one made. That is why I was getting `Connection ID not found`
2. Until recently (https://github.com/espressif/esp-nimble/commit/4486eee186023a51426401178bddce9a141eafdb), the callback argument of the new connection was set to the new connection object, instead of preserving the previous argument. Thus why I was getting odd m_conn_id values.

Seems like esp-nimble-cpp currently does not support `CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT=y`. Trying to support it does seem to be problematic since there are two distinct behavior depending on esp-nimble version.

For recent versions that correctly preserve the callback argument, we should be able to update `pClient->m_conn_id` from `event->reattempt_cnt.conn_handle` when handling the `BLE_GAP_EVENT_REATTEMPT_COUNT` event.

For prior versions that uses the wrong callback argument, it is much trickier. We probably have to retrieve the correct `NimBLEClient` object by matching with `peer_addr`, update its m_conn_id, and then update the callback argument.

If the above idea seems feasible (and I'm not misunderstanding the problem), I can try to whip up a patch.

h2zero, 2024-07-30:
Wow, thanks for doing the deep dive! I had not tried the connection reattempt myself, not surprised that it's causing issues here :(. Do you find using to be beneficial? I just thought it was easy enough to call `connect()` again if the first returned false. 

A PR would be very welcome for sure!

xieliwei, 2024-07-30:
Ugh, I've been thinking about how best to approach this fix and I can't arrive at a satisfactory solution. You're right that it isn't very useful. I have `CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT` turned on hoping to reduce some responsibilities in my application code, but now that I'm here, the irony is that it has caused more headache than time savings. =(

Anyway here are my justifications for probably #nofix-ing this:

1. There's no reliable way, even with pre-processor directives, to detect which esp-nimble version (with or without the overwritten callback) is in use
2. Turns out we can't recover the `NimBLEClient` object by matching with `peer_addr`. Doing so assumes that `peer_addr` is unique among all existing clients. However, there is nothing stopping existing esp-nimble-cpp users from having multiple clients with the same `peer_addr` (not sure what could be the reason to do so, but it might exist), so doing this might cross wires.
3. Unfortunately even for the case when the callback is preserved, I was wrong that we could update `pClient->m_conn_id` from `BLE_GAP_EVENT_REATTEMPT_COUNT`. That event is still fired from the old connection. Instead, for every event type, we have to check the event's `conn_handle` (if it is available) against `pClient->m_conn_id` and update it if they differ.

Except for point 1, we can actually overcome point 2 and 3 with kludgy code, but I don't think it is worth integrating that kludge for a rarely-used, somewhat experimental feature. It might actually be more beneficial fixing this on esp-nimble instead. I'll bug them when I get the time.

For now, I think there's enough information for anyone who has this issue to either:
1. Resolve it by turning `CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT` off, or
2. If support for reattempt is really desired, come up with their own workaround that suits their application

So we can close this issue if you're ok with this outcome.

h2zero, 2024-07-31:
Thanks for the insight, my own quick scan of it resulted in the same conclusion, not worth the effort to support. I'm not a big fan of the option they created with this as it seemed to be a minor thing for the application to do the same thing and have control over it. 

Espressif have added a lot of things like this recently, which have been somewhat problematic for this library. I think not supporting these additions by espressif, that haven't been merged upstream, may be the policy going forward as I use this library for other mcus as well and don't want to deal with the complexity.

For this situation I think replicating espressifs code by calling `connect()` again after failure is a pretty small addition to the application code.

xieliwei, 2024-08-03:
Yeah. Unfortunately I think all these changes probably have to do with eventual support for all the BT5.3/5.4 minutiae that upstream hasn't officially supported yet. Doesn't help that they haven't made a stance about doing a full fork or maintaining compatibility with upstream, so we see odd implementations that try to workaround how the upstream stack works, while diverging significantly for some features.

1.6.0 is due to be released this week and I wonder if we'll see things sync up again or we'll finally see Espressif decide to fully separate from upstream.

For now I'll close this issue. Thank you very much for your feedback and time on this!

xieliwei, 2024-08-03:
Whoops, probably should close with won't fix.

PR #230 (2024-11-17): Workaround for bug in NimBLE stack when connect re-attempt enabled.
Connect reattempt does not pass the arg parameter correctly and the disconnect event will have no client pointer in the argument. This finds the client from the connection handle to compensate.

Fixes #189
closed 2024-11-17 by h2zero