h2zero/esp-nimble-cpp #193: nimble GATT operation failed for unknown reason

PandaBalls, 2024-08-16:
I run the BLE server on my little esp32 c3 device with NimBLE,  and I build the web app with vue.
It works fine in Desktop Edge Browser: setValue / ReadValue / Receive Notify from esp device.

But on my Android Phone(Samsung), the Edge can bond with device, can setValue or ReadValue. But can not receive the notify from ESP32's NimBLE service. the mobile phone's Edge Console said:
`GATT operation failed for unknown reason`

PandaBalls, 2024-08-16:
more info from Edge Console:
[image]

h2zero, 2024-08-19:
No idea why this would be a problem for you, hopefully someone who is more familiar with android can assist.