h2zero/esp-nimble-cpp #200: setValue(std::vector<uint8_t>&) fails

finger563, 2024-10-01:
`NimBLEAttrValue::setValue(std::vector<uint8_t> s)` fails, likely because it is casting `v` improperly, using `reinterpret_cast<const uint8_t*>(&s)`.

I found this out because my HID code (where I manually create the characteristics was using this API, though it is disabled on the `NimBLEHIDDevice` class. If you re-enable it (as I have mentioned in the code [here](https://github.com/finger563/nimble-cpp-hid-issue-example/blob/main/main/main.cpp#L91)), then the call to setValue ends up pushing garbage data into the attribute value.

I'd suggest having a code path which checks if the type / container has a `data()` method, similarly to how you check the `c_str()` method, so that you can properly access the data for the container since right now the code will compile with std::vector, but will break.

h2zero, 2024-10-02:
I think this can be fixed by adding `[0]` after `&s` here:
`reinterpret_cast<const uint8_t*>(&s)`

finger563, 2024-10-02:
That wouldn't work since the use of `sizeof(T)` would be incorrect. I think there needs to a meta-case which handles the data/size, falling back to c_str/length if that's not available, and falling back to cast/sizeof if neither is available.

PR #201 (2024-10-14): fix: #200 Enable use of `data()`/`size()` before trying `c_str()`/`length()`
Closes #200 

Replace test for `c_str()` and `length()` (which only matches a few stl containers), for the more general `data()` and `size()` check.

I tested with this example application and it now works properly: 

https://github.com/finger563/nimble-cpp-hid-issue-example/tree/main

Note: I realize I accidentally committed the update to `NimBLEHIDDevice` to expose the `reportMap` characteristic. Personally I think it's a good change, but LMK if you want to revert that.
commit 3820f570 (2024-10-14): fix: #200 Enable use of `data()`/`size()` before trying `c_str()`/`length()` (#201)

* fix: #200 Use `data()`/`size()` instead of `c_str()`/`length()`

* Reduce duplication, only allow template function in characteristic and remote value attr if the type is not a pointer (otherwise sizeof is useless). add appropriate notes

* clean up AttValue::setValue to remove unnecessary length parameter enabling requirement of non-pointer type
closed 2024-10-14 by h2zero