h2zero/esp-nimble-cpp #202: Include error on new project

jefflongo, 2024-10-01:
Created a minimal project using esp-idf using Arduino as an IDF component following [these instructions](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html). Using arduino-esp32 3.0.5 and IDF 5.1.4. Then added NimBLE as a component to the project. This results in the following error:

```
fatal error: esp32-hal-bt.h: No such file or directory
```

I was able to make it compile by adding `arduino-esp32` to `ESP_NIMBLE_PRIV_REQUIRES` in the nimble CMakeLists.txt. I am not using platformio.

jefflongo, 2024-10-01:
If it helps, here are the contents of `BUILD_COMPONENTS`:
```
riscv;esp_ringbuf;efuse;driver;esp_pm;mbedtls;esp_app_format;bootloader_support;bootloader;esptool_py;partition_table;esp_partition;app_update;esp_mm;spi_flash;pthread;esp_system;esp_rom;hal;log;heap;soc;esp_hw_support;freertos;newlib;cxx;esp_common;esp_timer;app_trace;esp_event;nvs_flash;esp_phy;vfs;lwip;esp_netif_stack;esp_netif;wpa_supplicant;esp_coex;esp_wifi;bt;unity;cmock;console;http_parser;esp-tls;esp_adc;esp_eth;esp_gdbstub;esp_hid;tcp_transport;esp_http_client;esp_http_server;esp_https_ota;esp_https_server;esp_psram;esp_lcd;protobuf-c;protocomm;esp_local_ctrl;espcoredump;wear_levelling;sdmmc;fatfs;idf_test;ieee802154;json;mqtt;openthread;spiffs;ulp;usb;wifi_provisioning;espressif__network_provisioning;chmorgan__esp-libhelix-mp3;espressif__esp-modbus;espressif__libsodium;espressif__mdns;espressif__esp-dsp;espressif__esp-zboss-lib;espressif__esp-zigbee-lib;espressif__rmaker_common;espressif__esp_diagnostics;espressif__cbor;espressif__esp_diag_data_store;espressif__esp_insights;espressif__esp_modem;espressif__jsmn;espressif__json_parser;espressif__json_generator;espressif__esp_schedule;espressif__esp-serial-flasher;espressif__esp_rcp_update;espressif__esp_secure_cert_mgr;espressif__esp_rainmaker;espressif__qrcode;joltwallet__littlefs;espressif__arduino-esp32;main;esp-nimble-cpp
```

and `__hack_component_targets`:
```  
___idf_app_trace;___idf_app_update;___idf_bootloader;___idf_bootloader_support;___idf_bt;___idf_cmock;___idf_console;___idf_cxx;___idf_driver;___idf_efuse;___idf_esp-tls;___idf_esp_adc;___idf_esp_app_format;___idf_esp_coex;___idf_esp_common;___idf_esp_eth;___idf_esp_event;___idf_esp_gdbstub;___idf_esp_hid;___idf_esp_http_client;___idf_esp_http_server;___idf_esp_https_ota;___idf_esp_https_server;___idf_esp_hw_support;___idf_esp_lcd;___idf_esp_local_ctrl;___idf_esp_mm;___idf_esp_netif;___idf_esp_netif_stack;___idf_esp_partition;___idf_esp_phy;___idf_esp_pm;___idf_esp_psram;___idf_esp_ringbuf;___idf_esp_rom;___idf_esp_system;___idf_esp_timer;___idf_esp_wifi;___idf_espcoredump;___idf_esptool_py;___idf_fatfs;___idf_freertos;___idf_hal;___idf_heap;___idf_http_parser;___idf_idf_test;___idf_ieee802154;___idf_json;___idf_linux;___idf_log;___idf_lwip;___idf_mbedtls;___idf_mqtt;___idf_newlib;___idf_nvs_flash;___idf_openthread;___idf_partition_table;___idf_perfmon;___idf_protobuf-c;___idf_protocomm;___idf_pthread;___idf_riscv;___idf_sdmmc;___idf_soc;___idf_spi_flash;___idf_spiffs;___idf_tcp_transport;___idf_touch_element;___idf_ulp;___idf_unity;___idf_usb;___idf_vfs;___idf_wear_levelling;___idf_wifi_provisioning;___idf_wpa_supplicant;___idf_xtensa;___idf_main;___idf_esp-nimble-cpp;___idf_chmorgan__esp-libhelix-mp3;___idf_espressif__arduino-esp32;___idf_espressif__cbor;___idf_espressif__esp-dsp;___idf_espressif__esp-modbus;___idf_espressif__esp-serial-flasher;___idf_espressif__esp-zboss-lib;___idf_espressif__esp-zigbee-lib;___idf_espressif__esp_diag_data_store;___idf_espressif__esp_diagnostics;___idf_espressif__esp_insights;___idf_espressif__esp_modem;___idf_espressif__esp_rainmaker;___idf_espressif__esp_rcp_update;___idf_espressif__esp_schedule;___idf_espressif__esp_secure_cert_mgr;___idf_espressif__jsmn;___idf_espressif__json_generator;___idf_espressif__json_parser;___idf_espressif__libsodium;___idf_espressif__mdns;___idf_espressif__network_provisioning;___idf_espressif__qrcode;___idf_espressif__rmaker_common;___idf_joltwallet__littlefs
```

finger563, 2024-10-02:
@jefflongo I don't use the arduino framework, but does it automatically configure the sdkconfig for you? If not, you may need to enable bluetooth using `menuconfig` or manually within the `sdkconfig`, otherwise the bluetooth files will not be able to be found for compilation.

jefflongo, 2024-10-02:
@finger563 No, adding Arduino as a component does not automatically configure `sdkconfig` for you. I have enabled bluetooth in `sdkconfig`, but that is not the problem. The problem is that the `esp-nimble-cpp` component is not depending on `arduino-esp32`, which is how it inherits the include path for `esp32-hal-bt.h` which is part of the `arduino-esp32` core.

h2zero, 2024-10-02:
The build system seems to have changed and can no longer add the dynamic dependency for Arduino-esp32 via the cmake check 😞.

jefflongo, 2024-10-03:
@h2zero So is the solution to look for `espressif__arduino-esp32` in `BUILD_COMPONENTS` or `___idf_espressif__arduino-esp32` in `__hack_component_targets` and use that to add `arduino-esp32` to `ESP_NIMBLE_PRIV_REQUIRES`?

I don't know the context of the existing structure, but if that's all it is I can post a PR.

closed 2024-10-07 by h2zero

redstonee, 2024-10-28:
Same problem, but it happend after #204 is applied, it works fine if I checkout the commit before this PR.  
My Arduino component is installed by git submodule.

jefflongo, 2024-10-28:
Can you provide a minimum example to build a project with this error?

redstonee, 2024-10-29:
Sure, here is how I configure the project:
* Create a project with *arduino-as-componet* template ( IDF v5.1.4 )
* Change `configTICK_RATE_HZ` to 1000
* Enable `CONFIG_BT_ENABLED`
* Enable `CONFIG_BT_NIMBLE_ENABLED`
* Save sdkconfig
* Add Arduino framework by
 ```
git submodule add --depth 1 https://github.com/espressif/arduino-esp32.git components/arduino
```
* Add nimble library by
 ```
git submodule add https://github.com/h2zero/esp-nimble-cpp components/nimble
```
* Create an initial commit
 ```
git add .
git commit -m "Initial commit"
```
* Build the project, and the error occurs:
```
[1445/1667] Building CXX object esp-idf/nimble/CMakeFiles/__idf_nimble.dir/src/NimBLEDevice.cpp.obj
FAILED: esp-idf/nimble/CMakeFiles/__idf_nimble.dir/src/NimBLEDevice.cpp.obj 
ccache C:\Users\RedStone\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -DESP_PLATFORM -DIDF_VER=\"v5.1.4-dirty\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/RedStone/src/idf/tmp/build/config -IC:/Users/RedStone/src/idf/tmp/components/nimble/src -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/newlib/platform_include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/freertos/FreeRTOS-Kernel/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/freertos/esp_additions/include/freertos -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/freertos/esp_additions/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/freertos/esp_additions/arch/xtensa/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_hw_support/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_hw_support/include/soc -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_hw_support/include/soc/esp32 -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_hw_support/port/esp32/. -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/heap/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/log/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/soc/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/soc/esp32 -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/soc/esp32/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/hal/esp32/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/hal/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/hal/platform_port/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_rom/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_rom/include/esp32 -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_rom/esp32 -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_common/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_system/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_system/port/soc -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/esp_system/port/include/private -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/xtensa/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/xtensa/esp32/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/lwip/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/lwip/include/apps -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/lwip/include/apps/sntp -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/lwip/lwip/src/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/lwip/port/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/lwip/port/freertos/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/lwip/port/esp32xx/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/lwip/port/esp32xx/include/arch -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/include/esp32/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/common/osi/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/common/api/include/api -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/common/btc/profile/esp/blufi/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/common/btc/profile/esp/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/include 
-ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/hr/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/htp/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/prox/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/cts/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include 
-ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/hid/include -ID:/SDKs/ESP-idf/v5.1.4/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/sps/include -ID:/ [... cut ...] les/__idf_nimble.dir/src/NimBLEDevice.cpp.obj -MF esp-idf\nimble\CMakeFiles\__idf_nimble.dir\src\NimBLEDevice.cpp.obj.d -o esp-idf/nimble/CMakeFiles/__idf_nimble.dir/src/NimBLEDevice.cpp.obj -c C:/Users/RedStone/src/idf/tmp/components/nimble/src/NimBLEDevice.cpp
C:/Users/RedStone/src/idf/tmp/components/nimble/src/NimBLEDevice.cpp:54:12: fatal error: esp32-hal-bt.h: No such file or directory       
   54 | #  include "esp32-hal-bt.h"
      |            ^~~~~~~~~~~~~~~~
compilation terminated.
```

jefflongo, 2024-10-29:
Could you provide some more context? Here's what I tried. It's largely based off of [this workflow](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html).

- Using IDF v5.1.4
- Created project from template
```
idf.py create-project-from-example "espressif/arduino-esp32^3.0.2:hello_world"
```
- Add arduino as submodule
```
git submodule add --depth 1 https://github.com/espressif/arduino-esp32.git hello_world/components/arduino
git submodule add https://github.com/h2zero/esp-nimble-cpp hello_world/components/nimble
```
- Create initial commit
```
git add .
git commit -m "Initial commit"
```
- Enable `CONFIG_BT_ENABLED` and `CONFIG_BT_NIMBLE_ENABLED`
- Project builds successfully

redstonee, 2024-10-30:
Freaking amazing!  
I'm using the esp-idf VS Code extension for creating and building the project.
Besides I can figure out no difference between what we did.  
It's even more magical that I can use it normally without #204 merged. 🙃

jefflongo, 2024-10-30:
Can you edit the nimble CMakeLists.txt to print the contents of `BUILD_COMPONENTS` and `__hack_component_targets` like I did at the top of this issue?

redstonee, 2024-11-02:
Good idea!  
Here's my `BUILD_COMPONENTS` :
```
xtensa;esp_ringbuf;efuse;driver;esp_pm;mbedtls;esp_app_format;bootloader_support;bootloader;esptool_py;partition_table;esp_partition;app_update;esp_mm;spi_flash;pthread;esp_system;esp_rom;hal;log;heap;soc;esp_hw_support;freertos;newlib;cxx;esp_common;esp_timer;app_trace;esp_event;nvs_flash;esp_phy;vfs;lwip;esp_netif_stack;esp_netif;wpa_supplicant;esp_coex;esp_wifi;bt;unity;cmock;console;http_parser;esp-tls;esp_adc;esp_eth;esp_gdbstub;esp_hid;tcp_transport;esp_http_client;esp_http_server;esp_https_ota;esp_https_server;esp_psram;esp_lcd;protobuf-c;protocomm;esp_local_ctrl;espcoredump;wear_levelling;sdmmc;fatfs;idf_test;ieee802154;json;mqtt;openthread;perfmon;spiffs;ulp;usb;wifi_provisioning;main;espressif__network_provisioning;chmorgan__esp-libhelix-mp3;espressif__esp-modbus;espressif__libsodium;espressif__mdns;joltwallet__littlefs;espressif__esp-dsp;espressif__esp-zboss-lib;espressif__esp-zigbee-lib;espressif__rmaker_common;espressif__esp_diagnostics;espressif__cbor;espressif__esp_diag_data_store;espressif__esp_insights;espressif__esp_modem;espressif__jsmn;espressif__json_parser;espressif__json_generator;espressif__esp_schedule;espressif__esp-serial-flasher;espressif__esp_rcp_update;espressif__esp_secure_cert_mgr;espressif__esp_rainmaker;espressif__qrcode;arduino;nimble
```
`__hack_component_targets` : 
```
___idf_app_trace;___idf_app_update;___idf_bootloader;___idf_bootloader_support;___idf_bt;___idf_cmock;___idf_console;___idf_cxx;___idf_driver;___idf_efuse;___idf_esp-tls;___idf_esp_adc;___idf_esp_app_format;___idf_esp_coex;___idf_esp_common;___idf_esp_eth;___idf_esp_event;___idf_esp_gdbstub;___idf_esp_hid;___idf_esp_http_client;___idf_esp_http_server;___idf_esp_https_ota;___idf_esp_https_server;___idf_esp_hw_support;___idf_esp_lcd;___idf_esp_local_ctrl;___idf_esp_mm;___idf_esp_netif;___idf_esp_netif_stack;___idf_esp_partition;___idf_esp_phy;___idf_esp_pm;___idf_esp_psram;___idf_esp_ringbuf;___idf_esp_rom;___idf_esp_system;___idf_esp_timer;___idf_esp_wifi;___idf_espcoredump;___idf_esptool_py;___idf_fatfs;___idf_freertos;___idf_hal;___idf_heap;___idf_http_parser;___idf_idf_test;___idf_ieee802154;___idf_json;___idf_linux;___idf_log;___idf_lwip;___idf_mbedtls;___idf_mqtt;___idf_newlib;___idf_nvs_flash;___idf_openthread;___idf_partition_table;___idf_perfmon;___idf_protobuf-c;___idf_protocomm;___idf_pthread;___idf_riscv;___idf_sdmmc;___idf_soc;___idf_spi_flash;___idf_spiffs;___idf_tcp_transport;___idf_touch_element;___idf_ulp;___idf_unity;___idf_usb;___idf_vfs;___idf_wear_levelling;___idf_wifi_provisioning;___idf_wpa_supplicant;___idf_xtensa;___idf_main;___idf_arduino;___idf_nimble;___idf_chmorgan__esp-libhelix-mp3;___idf_espressif__cbor;___idf_espressif__esp-dsp;___idf_espressif__esp-modbus;___idf_espressif__esp-serial-flasher;___idf_espressif__esp-zboss-lib;___idf_espressif__esp-zigbee-lib;___idf_espressif__esp_diag_data_store;___idf_espressif__esp_diagnostics;___idf_espressif__esp_insights;___idf_espressif__esp_modem;___idf_espressif__esp_rainmaker;___idf_espressif__esp_rcp_update;___idf_espressif__esp_schedule;___idf_espressif__esp_secure_cert_mgr;___idf_espressif__jsmn;___idf_espressif__json_generator;___idf_espressif__json_parser;___idf_espressif__libsodium;___idf_espressif__mdns;___idf_espressif__network_provisioning;___idf_espressif__qrcode;___idf_espressif__rmaker_common;___idf_joltwallet__littlefs
```
I notify that there is `arduino` in `BUILD_COMPONENTS`, however it seems not working.  
But in `__hack_component_targets`, there's one item `___idf_arduino` (beginning with 3 underlines instead of 2). After I added an underline there, it can be built successfully.😦

jefflongo, 2024-11-02:
Interesting.. I'm not sure why that should be necessary though since the "arduino" in `BUILD_COMPONENTS` check should be enough. IMO The `__hack_component_targets` check should be removed entirely..

If I directly make a CMakeLists.txt and set those two variables to what you've provided, it seems to select the correct case, even if I remove the `__hack_component_targets` check entirely.

If you want to upload your minimal example to GitHub I can try on my machine.

h2zero, 2024-11-03:
I have just updated the cmake file to remove the `__hack_component_targets` , it was only useful for a time when the cmake build was new.

redstonee, 2024-11-05:
I reviewed the CMake configuration output and found that there are 2 messages printing `BUILD_COMPONENTS` and `__hack_component_targets`. Between them the first `BUILD_COMPONENTS` is empty but the second one includes `arduino`, while both of the `__hack_component_targets` includes `___idf_arduino`.   

The first ones:
[image]

The second ones:
[image]

> Good idea! Here's my `BUILD_COMPONENTS` :
>
> ``` [...]

redstonee, 2024-11-05:
The reason why I could build the previous commits of this library is probably that `__hack_component_targets MATCHES "__idf_arduino"` was used instead of `IN_LIST`, which enabled it to match `___idf_arduino` that has one more underline🤔

PR #204 (2024-10-07): Fix check for arduino component
Fixes #202. Tested with IDF 5.1.4 / Arduino core 3.0.x and IDF 4.4.8 / Arduino core 2.0.x.