h2zero/esp-nimble-cpp #205: [esp32-p4] it may works with very small update

chegewara, 2024-10-07:
Hi, this is just to let you know.
I am working on some esp-idf project which is working well on S3, but i wish to make it compatible with P4 too.
I have it running now, but it requires just a bit of work to remove some esp32 API from bluedroid and to remove this header `esp_bt.h`.

Code works without this header and API in functions
```
    static void             setPower(esp_power_level_t powerLevel, esp_ble_power_type_t powerType=ESP_BLE_PWR_TYPE_DEFAULT);
    static int              getPower(esp_ble_power_type_t powerType=ESP_BLE_PWR_TYPE_DEFAULT);
```

In addition it is required to change options in menuconfig (most likely with Kconfig)
```
# Override some defaults so BT stack is enabled
# in this example

#
# BT config
#
CONFIG_BT_ENABLED=y
CONFIG_BT_CONTROLLER_DISABLED=y
CONFIG_BT_BLUEDROID_ENABLED=n
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_TRANSPORT_UART=n

#
# Enable ESP Hosted BT
# Used as VHCI transport between BT Host and Controller
#
CONFIG_ESP_ENABLE_BT=y
```
and to add those 2 components from registry
```
  espressif/esp_hosted:
    version: ~0.0.12
  espressif/esp_wifi_remote:
    version: ~0.3.0
```

I can help if you need assistance.

Thanks

h2zero, 2024-10-11:
Thanks @chegewara, I'll look into this shortly!

h2zero, 2024-10-12:
Okay, since you're at it, want to make a PR? 🙂

chegewara, 2024-10-21:
Done.

chegewara, 2024-11-29:
Hi,
P4 support has been broken by some PR since then.

h2zero, 2024-11-29:
Compile error or something else?

chegewara, 2024-11-29:
Yes, it's compile error.

h2zero, 2024-11-30:
Can you share the output? I guess we should add the p4 to the ci

h2zero, 2024-12-03:
@chegewara I ran a ci build here: https://github.com/h2zero/esp-nimble-cpp/actions/runs/12132266168/job/33825859878#step:4:1423

Looks like something might have changed upstream?