h2zero/esp-nimble-cpp #207: BLE Modem Sleep

michaelboeding, 2024-10-12:
Are there any easy ways to enable BLE modem sleep in this library like the example here for the idf https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/power_save/main/main.c ?

h2zero, 2024-10-12:
There is no API for power save at the moment.

michaelboeding, 2024-10-12:
Hey @finger563 thanks I'll try this! You just set it up by setting up something like the below? 

`c++

 // Configure Bluetooth modem sleep
    // Set low power options, using external crystal for timing
    esp_pm_config_esp32_t pm_config = {
        .max_freq_mhz = 80,
        .min_freq_mhz = 10,
        .light_sleep_enable = true,
    };
    ESP_ERROR_CHECK(esp_pm_configure(&pm_config));

    ESP_LOGI(TAG, "Bluetooth modem sleep enabled");

`

finger563, 2024-10-14:
couple notes:
1. min freq (if I remember correctly) is 40mhz 
2. You also need to configure the following:
  1. `CONFIG_BT_CTRL_MODEM_SLEEP=y`
  2. `CONFIG_BT_CTRL_MAIN_XTAL_PU_DURING_LIGHT_SLEEP=y`
  3. `CONFIG_PM_POWER_DOWN_CPU_IN_LIGHT_SLEEP=y`
  4. `CONFIG_FREERTOS_USE_TICKLESS_IDLE=y`
  5. `CONFIG_FREERTOS_IDLE_TIMEOUT_BEFORE_SLEEP=3` though you can adjust this value to be shorter or longer depending on your application code

You may also want to enable the external crystal:
1. `CONFIG_RTC_CLK_SRC_EXT_CRYS=y`
2. `CONFIG_BT_CTRL_LPCLK_SEL_EXT_32K_XTAL=y`
3. `CONFIG_BT_CTRL_SLEEP_MODE_EFF=1`
4. `CONFIG_BT_CTRL_SLEEP_CLOCK_EFF=3`

I'd recommend looking into the help text for all those menuconfig options and their other / related options as it will help explain more what they do and how you may need to configure it for your specific system.