h2zero/esp-nimble-cpp #212: Would it make sense to report own address according to the set address type?

Staars, 2024-10-30:
When I set the address type with `NimBLE::setOwnAddrType()` and change the address with `ble_hs_id_set_rnd()` I still get my public address reported with `NimBLEDevice::getAddress()`.

For my personal taste it would make more sense to get the random address.

The following change would do this, but as this is not a bug but a design decision I did not create a (then cleaned up) PR.

```
NimBLEAddress NimBLEDevice::getAddress() {
    ble_addr_t addr = {m_own_addr_type, 0};

    if(BLE_HS_ENOADDR == ble_hs_id_copy_addr(m_own_addr_type, addr.val, NULL)) {
        // NIMBLE_LOGD(LOG_TAG, "Public address not found, checking random");
        // addr.type = BLE_ADDR_RANDOM;
        // ble_hs_id_copy_addr(BLE_ADDR_RANDOM, addr.val, NULL);
        return NimBLEAddress(); // return blank to report error
    }

    return NimBLEAddress(addr);
} // getAddress
```

h2zero, 2024-10-31:
Hi @Staars, I agree, this would make more sense. It wasn't really a design decision but an oversight and I would certainly accept a PR for it.

Staars, 2024-10-31:
As you are working on a PR around NimBLEDevice with breaking changes anyway, you may consider to include this code change to: https://github.com/h2zero/esp-nimble-cpp/pull/198 (and modify it to your liking).

It is not an urgent issue for me.

h2zero, 2024-10-31:
Okay, I'll do that

h2zero, 2024-11-01:
Done: made a few improvements to the setting of the device address type while I was at it :) Let me know what you think if you have a chance.

Staars, 2024-11-01:
If I read correctly, the new behaviour regarding the address types would be:
Setting address type to 1 will check for existing random address and automatically generate a new address if needed (which is really random).
Getting the device address takes the own type into account.
We have more error checking. 

This would work und looks good to me with one little exception.

If I set an own random address before that and this goes wrong for whatever reason, that call to setOwnAddrType to 1 would automagically create a new one and report success although it went wrong int the context of the application. (usecases like OpenHaystack)

From the application side it could be convenient to have another method, like NimBLE::setOwnRandomAddr, which does all the needed checks in the background, activates this random address and returns the correct error result.
(BTW I would not recommend to add even more abstractions for esp_iface_mac_addr_set() and the likes)

h2zero, 2024-11-01:
> If I set an own random address before that and this goes wrong for whatever reason, that call to setOwnAddrType to 1 would automagically create a new one and report success although it went wrong int the context of the application. (usecases like OpenHaystack)

If you already set the address then you would know it went wrong and probably shouldn't call this function 😄. That said,  I think the random id address should be set when the stack starts so that the call here should never fail and the code to generate the address in this function would not be required.

> From the application side it could be convenient to have another method, like NimBLE::setOwnRandomAddr, which does all the needed checks in the background, activates this random address and returns the correct error result.
(BTW I would not recommend to add even more abstractions for esp_iface_mac_addr_set() and the likes)

I have planned to do this in a separate commit once this PR is ready.

h2zero, 2024-11-02:
The latest changes ensure there is a random address set at startup and will not change the address when calling setOwnAddrType.

Now to add the setOwnAddr function and this should be good to go.

PR #198 (2024-11-03): [BREAKING] Refactor NimBLEDevice
* General code cleanup
* `NimBLEDevice::getInitialized` renamed to `NimBLEDevice::isInitialized`.
* `NimBLEDevice::setPower` no longer takes the `esp_power_level_t` and `esp_ble_power_type_t`, instead only an integer value in dbm units is accepted.
* `NimBLEDevice::setPower` now returns a bool value, true = success.
* `NimBLEDevice::setMTU` now returns a bool value, true = success.
* `NimBLEDevice::getAddress` will now return the address currently in use.
* `NimBLEDevice::init` now returns a bool with `true` indicating success.
* `NimBLEDevice::deinit` now retu [... cut]
closed 2024-11-03 by h2zero