h2zero/esp-nimble-cpp #216: Advertising slow an ESP32C3 but very fast on ESP32

0FR0, 2024-11-07:
Hi,
i am i having an issue with the esp32c3 and BLE advertising. 
The delay between packages 64ms and a lot of seconds (+ 13 sec).
When i put the same code on a esp32 the delay between packages is 0 to 200ms.

I tryed to set the delay on the esp32c3 code but it made it worse.
Does any one else had seen the same problem or any idea how to fix it?

Sender code:
`#include <NimBLEDevice.h>

extern "C"
{
  void app_main(void);
}

void app_main(void)
{
  BLEDevice::init("Techniker Projekt 2024");
  BLEDevice::startAdvertising();
}`

h2zero, 2024-11-09:
How are you setting the interval?

0FR0, 2024-11-10:
i tryed:
```
...
BLEDevice::init("Techniker Projekt 2024");
BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
pAdvertising->setMaxInterval(32);
pAdvertising->setMinInterval(32);
pAdvertising->setMaxPreferred(32);
pAdvertising->setMinPreferred(32);
BLEDevice::startAdvertising();
...
```
with different values.

h2zero, 2024-11-16:
Not sure about this one yet but I'm seeing some similarly odd behavior with the scanning start/stop. Will be investigating