h2zero/esp-nimble-cpp #220: how to import this component?

vhp8rc7p, 2024-11-11:
i tried to clone this repo and extract it to components folder in my project,it can't find NImBLEDevice.h when i try to build,then i modified idf_component.yml to this:
## IDF Component Manager Manifest File
dependencies:
  idf: ">=4.4"
  usb_host_hid: "^1.0.1"
  esp-nimble-cpp:
    override_path: "../components/esp-nimble-cpp"
    version: "*"
then when i try to build there are a ton of errors
[image]
i'm using espidf 5.3.1 and set target to esp32s3,enabled bt in menuconfig,set host to nimble,
i'm new to espidf,any help will be appreciated

h2zero, 2024-11-11:
Start with a clean project then make a folder called `components` then clone this repo into that folder. Now copy/paste one of the examples into your project folder and enable BT and NimBLE in menuconfig, it should build fine from there.

vhp8rc7p, 2024-11-13:
> Start with a clean project then make a folder called `components` then clone this repo into that folder. Now copy/paste one of the examples into your project folder and enable BT and NimBLE in menuconfig, it should build fine from there.

thank u for your help,i started a fresh new project, turned on bluetooth in menuconfig and set host to nimble,the error still persists,i guess there is something wrong with what i did,i will try to figure out

closed 2024-11-13 by vhp8rc7p