h2zero/esp-nimble-cpp #232: NimBLEClient::getRssi() returning 0 on async connection

thekurtovic, 2024-11-18:
Printing `NimBLEClient::getRssi` within `onConnect ` for async connection tends to always show 0*. No issue getting RSSI value after non-async connection is made. Oddly there's no error given as `ble_gap_conn_rssi` is returning 0.
Testing with https://github.com/h2zero/esp-nimble-cpp/pull/214

*Occasionally will get a non-zero value but it's rare.

Async
```
D (8884) NimBLEScan: << stop()
D (8884) NimBLEClient: >> connect(e3:e4:5b:75:ad:36)
D (9846) NimBLEClient: Got Client event
I (9846) NimBLEClient: Connected event
I [  9847][main.cpp:144] Connected to e3e45b75ad36, RSSI 0 (bond 1)
D (9862) NimBLEClient: Got Client event
I (9863) NimBLEClient: mtu update: mtu=247
D (9873) NimBLEClientCallbacks: onMTUChange: default
D (9873) NimBLEClient: exchangeMTUCb: status=0, mtu=247
D (9943) NimBLEClient: Got Client event
I (9944) NimBLEClient: Connection parameters updated.
D (9952) NimBLEClient: Got Client event
D (9975) NimBLEClient: Got Client event
D (10297) NimBLEClient: Got Client event
D [ 10297][main.cpp:182] Auth event 1 1 1 16
```

Non-async
```
D (13520) NimBLEScan: << stop()
D (13520) NimBLEScan: erase device: e3:e4:5b:75:ad:36
I [ 13511][main.cpp:354] Create client, attempting to connect...
D (13531) NimBLEClient: >> connect(e3:e4:5b:75:ad:36)
D (15487) NimBLEClient: Got Client event
I (15487) NimBLEClient: Connected event
D (15502) NimBLEClient: Got Client event
I (15503) NimBLEClient: mtu update: mtu=247
D (15503) NimBLEClientCallbacks: onMTUChange: default
D (15513) NimBLEClient: exchangeMTUCb: status=0, mtu=247
D (15513) NimBLEClient: Got Client event
I (15515) NimBLEClient: Connection established
I [ 15524][main.cpp:144] Connected to e3e45b75ad36, RSSI -40 (bond 1)
D (15540) NimBLEClient: Got Client event
D (15540) NimBLEClient: << connect()
I [ 15552][main.cpp:371] Securing connection...
D (15552) NimBLEClient: >> secureConnection()
D (15614) NimBLEClient: Got Client event
I (15614) NimBLEClient: Connection parameters updated.
D (15652) NimBLEClient: Got Client event
D [ 15652][avx_ble_cb.cpp:182] Auth event 1 1 1 16
```

h2zero, 2024-11-18:
Quite curious, probably an upstream issue as I don't see any error response in the log.

h2zero, 2024-11-18:
Just did some testing here, not able to reproduce it.

thekurtovic, 2024-11-18:
I tried the `NimBLE Async Client` example and still see the issue. `Advanced/NimBLE Client` works as expected.
I'm testing with esp32-s3 and idf 5.1.4.

sdkconfig.defaults
```
# NimBLE Options
CONFIG_BT_NIMBLE_NVS_PERSIST=y
CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL=y
CONFIG_BT_NIMBLE_LOG_LEVEL_NONE=y
# CONFIG_BT_NIMBLE_LOG_LEVEL_INFO is not set
CONFIG_BT_NIMBLE_LOG_LEVEL=4
CONFIG_BT_NIMBLE_MAX_BONDS=128
CONFIG_BT_NIMBLE_MAX_CCCDS=128
CONFIG_BT_NIMBLE_NVS_PERSIST=y
CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT=n
CONFIG_BT_NIMBLE_ATT_PREFERRED_MTU=247
CONFIG_BT_NIMBLE_EXT_ADV=y
CONFIG_BT_NIMBLE_MAX_EXT_ADV_INSTANCES=1
CONFIG_BT_NIMBLE_EXT_ADV_MAX_SIZE=1650
# CONFIG_BT_NIMBLE_ENABLE_PERIODIC_ADV is not set

CONFIG_BT_CTRL_DFT_TX_POWER_LEVEL_P20=y
CONFIG_BT_CTRL_DFT_TX_POWER_LEVEL_EFF=15

# BT config
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
CONFIG_BTDM_CTRL_MODE_BTDM=n
CONFIG_BT_BLUEDROID_ENABLED=n
CONFIG_BT_NIMBLE_ENABLED=y
```

h2zero, 2024-11-18:
Try with an original esp32, its working for me with the same idf version

thekurtovic, 2024-11-19:
I'll order one and see how it goes

thekurtovic, 2024-11-25:
Not sure if it's intentional but the `onConnect` callback is called even if an async connection fails. I noticed I was also getting 0xFFFF for the connection handle when RSSI was showing 0.
https://github.com/h2zero/esp-nimble-cpp/blob/master/src/NimBLEClient.cpp#L977-L999
I check for a valid handle now before processing the connection to avoid assuming success.

Also I haven't been getting the 0 RSSI readings on master anymore after fixing the above issue, even though they were separate issues.

h2zero, 2024-11-25:
Thanks, yes, I've been thinking about how best to handle the connection failure for async mode. I'm leaning toward a different callback, `onConnectFailure` or something similar. Thoughts?

thekurtovic, 2024-11-25:
Sounds good to me, that would have been my suggestion.

closed 2024-11-25 by thekurtovic