h2zero/esp-nimble-cpp #234: Can't compile on IDF 5.3

thekurtovic, 2024-11-19:
Wanted to start testing on 5.3 but I get errors compiling unless I go back to 68b82f5b855040b320c7865153e6289904ef132f.
Build output is really long, I can't make any sense of it.
[output.txt](https://github.com/user-attachments/files/17819081/output.txt)

Testing with this example on esp32-s3.
https://github.com/pioarduino/platform-espressif32/tree/develop/examples/espidf-arduino-h2zero-BLE_scan

5.3 isn't part of the workflow checks so I'm not sure if it's supported yet.

h2zero, 2024-11-19:
Try adding:
```
/****  FIX COMPILATION ****/
# undef min
# undef max
/**************************/
```
To here: https://github.com/h2zero/esp-nimble-cpp/blob/5f2730de023ddba98edf4b3e06d9c4c8ad61dde5/src/NimBLEUUID.h#L26

thekurtovic, 2024-11-20:
https://github.com/h2zero/esp-nimble-cpp/pull/236 resolves the issue.
Thank you

PR #236 (2024-11-26): Fix compilation errors in some situations.
Fixes #234
closed 2024-11-26 by h2zero