h2zero/esp-nimble-cpp #259: NimBLEHIDDevice can't reconnect after disconnect

obertake, 2024-12-10:
Hi h2zero,

I've ran into this issue where if I disconnect from a NimBLEHIDDevice, it won't reconnect (actually seeing different behavior for different devices (explained below)).

Chip: ESP32C6
ESP-IDF version: I've tried v5.1.4, v5.3, and v5.3, all exhibit the same behavior.
esp-nimble-cpp version: I've tried v1.4.1 as well as latest in master (as of yesterday), and also tried both of these in all of the aforementioned ESP-IDF versions.

Here is the small sample program that will build and have the issue:

```cpp
#include <NimBLEDevice.h>
#include <NimBLEHIDDevice.h>

#define HID_REPORT_ID 0x01

static const uint8_t HID_REPORT_DESC[] =
{
	// CONSUMER BUTTON
	USAGE_PAGE(1),		0x0C,			// USAGE_PAGE (Consumer)
	USAGE(1),		0x01,			// USAGE (Consumer Control)
	COLLECTION(1),		0x01,			// COLLECTION (Application)
	REPORT_ID(1),		HID_REPORT_ID,		// REPORT_ID (1)
	LOGICAL_MINIMUM(1),	0x00,			// LOGICAL_MINIMUM (0)
	LOGICAL_MAXIMUM(1),	0x01,			// LOGICAL_MAXIMUM (1)
	REPORT_SIZE(1),		0x01,			// REPORT_SIZE (1)
	REPORT_COUNT(1),	0x08,			// REPORT_COUNT (8)
	USAGE(1),		0xE9,			// Volume Increase
	USAGE(1),		0xEA,			// Volume Decrease
	USAGE(1),		0xE2,			// Volume Mute
	USAGE(1),		0x40,			// Menu
	USAGE(1),		0xCD,			// Play
	USAGE(1),		0xB7,			// Stop
	USAGE(1),		0xB5,			// Next Track
	USAGE(1),		0xB6,			// Prev Track
	HIDINPUT(1),		0x02,			// INPUT (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
	END_COLLECTION(0)				// END COLLECTION
};

extern "C" void app_main()
{
	printf("Starting app_main\n");

	NimBLEDevice::init("HIDTestDev");
	NimBLEDevice::setSecurityAuth(true, false, false);

	NimBLEServer *pServer = NimBLEDevice::createServer();

	NimBLEHIDDevice * pHidDevice = new NimBLEHIDDevice(pServer);
	pHidDevice->inputReport(HID_REPORT_ID);
	pHidDevice->reportMap((uint8_t*)HID_REPORT_DESC, sizeof(HID_REPORT_DESC));
	pHidDevice->startServices();

	NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
	pAdvertising->addServiceUUID(pHidDevice->hidService()->getUUID());
	pAdvertising->start();
}
```

Actual HID service code omitted because this is all we need to repro the issue.

Here is the behavior I'm seeing on the following devices:

iOS (iPhone SE):
- Pair device (first attempt always successful)
- Turn off ESP to disconnect
- Turn on ESP
- Notice that it doesn't automatically reconnect
- Tapping on connect manually (on phone) will attempt to connect, but always fail

WIndows PC:
- Pair device (first attempt always successful)
- Turn off ESP to disconnect
- Turn on ESP
- Pair will continuously try to reconnect and fail immediately (constantly flips between connected and disconnected in WIndows settings)

Samsung Galaxy S8:
- Pair device (first attempt always successful)
- Turn off ESP to disconnect
- Turn on ESP
- Notice that it doesn't automatically reconnect
- Turn off and on ESP again
- Will pair successfully on second attempt
- Alternates between a failed pairing and a successful pairing on each on/off cycle

Here is the monitor log for failed reconnect on Galaxy S8 (first pairing succeeds, then I turn off and on the ESP and reconnect fails)
[monitorlog_galaxys8.txt](https://github.com/user-attachments/files/18070256/monitorlog_galaxys8.txt)

obertake, 2024-12-10:
Also adding here the monitor log for Windows.
[monitorlog_windows.txt](https://github.com/user-attachments/files/18070715/monitorlog_windows.txt)

I am seeing lots of "BLE_GAP_EVENT_DISCONNECT reason=531" on Windows (but not other platforms)

obertake, 2024-12-10:
If I'm reading the error codes correctly, on `ble_hs.h`, error code 531 = 0x213, which translates to BLE_HS_ETIMEOUT with a BLE_HS_ERR_HCI_BASE mask (HCI component?)

obertake, 2024-12-10:
Also to note, in the logs you will see that I have Bluetooth modem sleep on, but turning it off has the same outcome.

obertake, 2024-12-10:
I also took a look at the windows bluetooth event logs to see what's going on.
It looks like there is an event for "attempting to pair to radio" but it doesn't have a success event.
When attempting a reconnect, it actually writes a "connection successfully established" event despite it immediately disconnecting and tries again.

[image]

[WindowsEventLog.txt](https://github.com/user-attachments/files/18071133/WindowsEventLog.txt)

obertake, 2024-12-10:
Also attaching my sdkconfig file here.
[sdkconfig.txt](https://github.com/user-attachments/files/18072016/sdkconfig.txt)

obertake, 2024-12-10:
Looking at the nimble logs, I did notice that the failure connections were not receiving a BLE_GAP_EVENT_MTU event after BLE_GAP_EVENT_CONNECT.

Here I'm attaching both the successful and unsuccessful connection logs from my Android and Windows tests.

[android_nimble_log_success.txt](https://github.com/user-attachments/files/18075493/android_nimble_log_success.txt)
[android_nimble_log_fail.txt](https://github.com/user-attachments/files/18075494/android_nimble_log_fail.txt)
[windows_nimble_log_success.txt](https://github.com/user-attachments/files/18075496/windows_nimble_log_success.txt)
[windows_nimble_log_fail.txt](https://github.com/user-attachments/files/18075498/windows_nimble_log_fail.txt)

h2zero, 2024-12-10:
Hello @obertake thank you for providing all of this information, it is very helpful.
I have tested the code you provided, adjusted to use current master commit:
```
#include <NimBLEDevice.h>
#include <NimBLEHIDDevice.h>
#include <HIDTypes.h>

#define HID_REPORT_ID 0x01

static const uint8_t HID_REPORT_DESC[] =
{
	// CONSUMER BUTTON
	USAGE_PAGE(1),		0x0C,			// USAGE_PAGE (Consumer)
	USAGE(1),		0x01,			// USAGE (Consumer Control)
	COLLECTION(1),		0x01,			// COLLECTION (Application)
	REPORT_ID(1),		HID_REPORT_ID,		// REPORT_ID (1)
	LOGICAL_MINIMUM(1),	0x00,			// LOGICAL_MINIMUM (0)
	LOGICAL_MAXIMUM(1),	0x01,			// LOGICAL_MAXIMUM (1)
	REPORT_SIZE(1),		0x01,			// REPORT_SIZE (1)
	REPORT_COUNT(1),	0x08,			// REPORT_COUNT (8)
	USAGE(1),		0xE9,			// Volume Increase
	USAGE(1),		0xEA,			// Volume Decrease
	USAGE(1),		0xE2,			// Volume Mute
	USAGE(1),		0x40,			// Menu
	USAGE(1),		0xCD,			// Play
	USAGE(1),		0xB7,			// Stop
	USAGE(1),		0xB5,			// Next Track
	USAGE(1),		0xB6,			// Prev Track
	HIDINPUT(1),		0x02,			// INPUT (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
	END_COLLECTION(0)				// END COLLECTION
};

extern "C" void app_main()
{
	printf("Starting app_main\n");

	NimBLEDevice::init("HIDTestDev");
	NimBLEDevice::setSecurityAuth(true, false, false);

	NimBLEServer *pServer = NimBLEDevice::createServer();

	NimBLEHIDDevice * pHidDevice = new NimBLEHIDDevice(pServer);
	pHidDevice->setManufacturer("DeviceManuTest");
	pHidDevice->setPnp(0x02, 0x0B0B, 0x0C0C, 0x0101);
	pHidDevice->setHidInfo(0x00, 0x01);
	pHidDevice->getInputReport(HID_REPORT_ID);
	pHidDevice->setReportMap((uint8_t*)HID_REPORT_DESC, sizeof(HID_REPORT_DESC));
	pHidDevice->startServices();

	NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
	pAdvertising->setAppearance(0x03C1);	//HID Keyboard icon
	pAdvertising->addServiceUUID(pHidDevice->getHidService()->getUUID());
	pAdvertising->start();
}
```

Using IDF v5.2.3 and an esp32c6 devkit I am not able to repro this issue. Tested with linux, windows and iphone and each device reliably reconnected after resetting the esp.

From the logs my guess is that you have a bad module or xtal on the board as the connection timeout error is commonly seen in that case (bad timing). If you have another device tot test with I suggest trying that to see if it makes a difference, best would be a completely different module/board or esp32 device.

obertake, 2024-12-10:
Hey h2zero, thanks for checking this one out. That's very odd indeed.
I've checked on a few other chips after I read your response, but no luck unfortunately.
Where can I drop you a message to arrange a paid consultation?

closed 2024-12-15 by obertake

luchidalgo, 2025-01-23:
Hello, did you find any solutions ?