h2zero/esp-nimble-cpp #270: Unhandled exception after authentication callback in NimBLEServer

finger563, 2024-12-19:
[image]

I've been trying to debug this for a little while and have not had much luck, so I figured I'd reach out to see if you've seen this.

I'm doing a sequence of 

adv -> connect / auth -> short delay -> disconnect -> short delay -> adv -> connect / auth -> disconnnect -> ...

on the server, where the connect is of course triggered by the BLE central to which this device is bonded. The delay is short, and is just me using buttons to inform the device to perform the action.

h2zero, 2024-12-19:
Are you changing characteristics or services in between? Looks like the client is subscribed to notifications of a now non-existent characteristic.

finger563, 2024-12-19:
I'm not changing the characteristics - but I am updating the value; I did notice that it doesn't crash if i comment out the `BLE_ATT_OP_NOTIFY_REQ` case in `nimble/host/src/ble_gatts.c` so it definitely seems like something may be going on there.

h2zero, 2024-12-19:
Looks like it was trying to call a function at address 0x0, but I'm unsure what that would be as the stack is good about checking callbacks.

finger563, 2024-12-19:
The only thing I'm doing is calling `NimBLEService::start()` on my services each time I connect. If I don't do that then it doesn't crash, but of course the functionality doesn't work after the first connect. Perhaps something in there is not clearing / restarting the service correctly?

h2zero, 2024-12-19:
You started advertising before starting the services?

finger563, 2024-12-19:
no, start the services before advertising each time

h2zero, 2024-12-19:
Oh okay, I thought you were starting the services when a client connects, which would certainly fail.

finger563, 2024-12-20:
As I'm looking through the changes, I've noticed that the `NimBLECharacteristic::sendValue` no longer performs any of the property or security checks that it used to - do you think that that could be part of the issue?

h2zero, 2024-12-20:
I think this one would benefit from some hardware debugging, setting a breakpoint right where the notify function is called and stepping through.

finger563, 2024-12-20:
Yea I'll try to get that setup. Unfortunately, the hardware I'm working on has no JTAG/UART access and the USB is being used by our code for other things.

finger563, 2024-12-20:
That being said, even if i simply return at the beginning of `setValue` the crash still occurs, so I do believe it's something else with the service / characteristic, or perhaps something else.

h2zero, 2024-12-20:
It's not your code that is calling the notify function, the stack is doing that when a bonded device reconnects. When the stack calls notify it calls the read characteristic function to get the data to notify and that seems to be where this crashes.

finger563, 2024-12-20:
In the `NimBLEService::start()` if I change it so that at the beginning:

```c++
bool NimBLEService::start() {
    NIMBLE_LOGD(LOG_TAG, ">> start(): Starting service: %s", toString().c_str());
    // If started previously, clear everything and start over
    if (m_pSvcDef->characteristics) {
        return true; // ADDED: do not modify the services if they exist already.
        if (m_pSvcDef->characteristics->descriptors) {
            delete[] m_pSvcDef->characteristics->descriptors;
        }
        delete[] m_pSvcDef->characteristics;
    }

... rest of the function continues below
```

Then the crash does not occur. I believe something with the destructor of these is not properly updating the nimble stack or is not properly locking the nimble stack before they become invalid.

h2zero, 2024-12-20:
Hmm good catch, `if (m_pSvcDef->characteristics) {` should probably be changed to if (m_pSvcDef->characteristics && getServer()->m_svcChanged) {`

finger563, 2024-12-20:
awesome, I'll test that and let you know!

finger563, 2024-12-20:
out of curiosity, would it make sense to keep the Characteristic changes i listed above or no?

finger563, 2024-12-20:
based on the rest of the function, i believe it should be something like:

```c++
bool NimBLEService::start() {
    NIMBLE_LOGD(LOG_TAG, ">> start(): Starting service: %s", toString().c_str());

    // If started previously and no characteristics have been added or removed,
    // then we can skip the service registration process.
    if (m_pSvcDef->characteristics && !getServer()->m_svcChanged) {
        return true;
    }

    // If started previously, clear everything and start over
    if (m_pSvcDef->characteristics) {
        if (m_pSvcDef->characteristics->descriptors) {
            delete[] m_pSvcDef->characteristics->descriptors;
        }
        delete[] m_pSvcDef->characteristics;
    }

    size_t numChrs = 0;
    for (const auto& chr : m_vChars) {
        if (chr->getRemoved()) {
            continue;
        }
        ++numChrs;
    }

    NIMBLE_LOGD(LOG_TAG, "Adding %d characteristics for service %s", numChrs, toString().c_str());
    if (numChrs) {
        int i = 0;

        // Nimble requires the last characteristic to have it's uuid = 0 to indicate the end
        // of the characteristics for the service. We create 1 extra and set it to null
        // for this purpose.
        ble_gatt_chr_def* pChrs = new ble_gatt_chr_def[numChrs + 1]{};
        for (const auto& chr : m_vChars) {
            if (chr->getRemoved()) {
                continue;
            }

            size_t numDscs = 0;
            for (const auto& dsc : chr->m_vDescriptors) {
                if (dsc->getRemoved()) {
                    continue;
                }
                ++numDscs;
            }

            if (numDscs) {
                int j = 0;

                // Must have last descriptor uuid = 0 so we have to create 1 extra
                ble_gatt_dsc_def* pDscs = new ble_gatt_dsc_def[numDscs + 1]{};
                for (const auto& dsc : chr->m_vDescriptors) {
                    if (dsc->getRemoved()) {
                        continue;
                    }

                    pDscs[j].uuid         = dsc->getUUID().getBase();
                    pDscs[j].att_flags    = dsc->getProperties();
                    pDscs[j].min_key_size = 0;
                    pDscs[j].access_cb    = NimBLEServer::handleGattEvent;
                    pDscs[j].arg          = dsc;
                    ++j;
                }

                pChrs[i].descriptors = pDscs;
            }

            pChrs[i].uuid         = chr->getUUID().getBase();
            pChrs[i].access_cb    = NimBLEServer::handleGattEvent;
            pChrs[i].arg          = chr;
            pChrs[i].flags        = chr->getProperties();
            pChrs[i].min_key_size = 0;
            pChrs[i].val_handle   = &chr->m_handle;
            ++i;
        }

        m_pSvcDef->characteristics = pChrs;
    }

    m_pSvcDef->type = BLE_GATT_SVC_TYPE_PRIMARY;
    int rc          = ble_gatts_count_cfg(m_pSvcDef);
    if (rc != 0) {
        NIMBLE_LOGE(LOG_TAG, "ble_gatts_count_cfg failed, rc= %d, %s", rc, NimBLEUtils::returnCodeToString(rc));
        return false;
    }

    rc = ble_gatts_add_svcs(m_pSvcDef);
    if (rc != 0) {
        NIMBLE_LOGE(LOG_TAG, "ble_gatts_add_svcs, rc= %d, %s", rc, NimBLEUtils::returnCodeToString(rc));
        return false;
    }

    NIMBLE_LOGD(LOG_TAG, "<< start()");
    return true;
} // start
```

finger563, 2024-12-20:
In my testing, this change fixes the crash

PR #271 (2024-12-20): Fix crash that could occur when server receives reconnection from client
* Update to not change services if they have not actually changed when starting the server 

Closes #270
commit e843c681 (2024-12-20): Fix crash that could occur when server reconnects
* Update to not change services if they have not actually changed when starting the server
Closes #270
commit fd0187bf (2024-12-22): Fix crash that could occur when server reconnects
* Update to not change services if they have not actually changed when starting the server
Closes #270
closed 2024-12-20 by h2zero