h2zero/esp-nimble-cpp #294: incorporating with esphome?

EasilyBoredEngineer, 2025-01-17:
Does anyone have any tangible examples of how to incorporate this with esphome?

h2zero, 2025-01-17:
Interesting question, not sure how I would go about it, but esphome has their own BLE functionality so it's never been brought up before.

EasilyBoredEngineer, 2025-01-17:
@h2zero - yep but their BLE implementation is really poor - very buggy, very memory intensive, very prone to random crashes - you can see a big discussion about it here - https://github.com/esphome/feature-requests/issues/810

h2zero, 2025-01-17:
I guess you could add this as a `lib_dep` in the esphome config then add a lambda or something in a startup hook, just my limited understanding. Otherwise a custom component may be needed.