h2zero/esp-nimble-cpp #296: Device is not persisting configured number of bonds

jefflongo, 2025-01-26:
Not sure if this is the right place to post this issue.. but I have a device with which I am bonding multiple devices.  I have enabled `CONFIG_BT_NIMBLE_NVS_PERSIST` and configured `CONFIG_BT_NIMBLE_MAX_BONDS` to 5, but it appears that when bonding a 4th device the ESP32 loses the encryption key of one of the other 3 devices and I must repair it. So it appears I can only maintain 3 encryption keys in NVS. Any idea what I could be doing wrong or is this a bug?

thekurtovic, 2025-01-26:
Just an idea, maybe adding a callback to check store status could provide some insight?
https://github.com/h2zero/esp-nimble-cpp/pull/295

Does it make any difference if you change max bonds/ccds to a higher value like 16?

h2zero, 2025-01-26:
You may have had an old bond still stored, try erasing the flash and start again.

jefflongo, 2025-01-26:
> You may have had an old bond still stored, try erasing the flash and start again.

Supposing this is the case, I would expect that the stale bonds get deleted first. When I start repairing the other devices I can see one of the other devices becomes unpaired. I can keep doing this and cycling through only 3 devices being paired.

thekurtovic, 2025-01-26:
Interesting, thanks for testing. I've seen my esp32's go beyond 3 bonds, strange that you are having an issue with it.
I take it there's no warnings/errors emitted from the controller?

FWIW here's where the event is originating
https://github.com/espressif/esp-nimble/blob/a4cd6027a544966d569db92c68e557f09e1f3d76/nimble/host/src/ble_store.c#L61-L71

Also forgot some documentation in my PR, but the callback should return either 0 or 1.
```
/**
 * Indicates an inability to perform a store operation.  This callback should
 * do one of two things:
 *     o Address the problem and return 0, indicating that the store operation
 *       should proceed.
 *     o Return nonzero to indicate that the store operation should be aborted.
 *
 * @param event                 Describes the store event being reported.
 * @param arg                   Optional user argument.
 *
 * @return                      0 if the store operation should proceed;
 *                              nonzero if the store operation should be
 *                                  aborted.
 */
```

jefflongo, 2025-01-26:
Doing some more testing.. I was able to pair all 4 after erasing the flash. I'm not sure why that step was necessary though or if I will run into this issue again at some later time.

> Interesting, thanks for testing. I've seen my esp32's go beyond 3 bonds, strange that you are having an issue with it. I take it there's no warnings/errors emitted from the controller?
>
> FWIW here's [...]

That explains that problem then, I was returning 0 so that probably just resulted in an infinite loop/undefined behavior.

thekurtovic, 2025-01-26:
Stupid question: you had already tried calling `NimBLEDevice::deleteAllBonds()` right?
Requiring a full erase is concerning

jefflongo, 2025-01-26:
No, I have not. TBH I didn't know that function even existed. Unfortunately now that I have erased the flash, I'm no longer able to currently reproduce the problem. I think I will close this issue and reopen if the issue comes up again at some later time.

closed 2025-01-26 by jefflongo