h2zero/esp-nimble-cpp #302: CMake Kconfig warnings when building with IDF 5.4

johnboiles, 2025-02-05:
I'm seeing these warnings:

```
warning: default on the choice symbol BT_NIMBLE_ENABLED (defined at /Users/johnboiles/Developer/repos/espressif/esp-idf-54/components/bt/Kconfig:22, /Users/johnboiles/Developer/repos/sindarin-inc/embedded-app/components/esp-nimble-cpp/Kconfig:217) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol BT_NIMBLE_ENABLED (defined at /Users/johnboiles/Developer/repos/espressif/esp-idf-54/components/bt/Kconfig:22, /Users/johnboiles/Developer/repos/sindarin-inc/embedded-app/components/esp-nimble-cpp/Kconfig:217) is defined with a prompt outside the choice
info: INFO: Symbol BT_ENABLED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /Users/johnboiles/Developer/repos/sindarin-inc/embedded-app/components/esp-nimble-cpp/Kconfig:210
    /Users/johnboiles/Developer/repos/espressif/esp-idf-54/components/bt/Kconfig:3
info: INFO: Symbol BT_NIMBLE_ENABLED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /Users/johnboiles/Developer/repos/espressif/esp-idf-54/components/bt/Kconfig:22
    /Users/johnboiles/Developer/repos/sindarin-inc/embedded-app/components/esp-nimble-cpp/Kconfig:217
info: INFO: Symbol BT_NIMBLE_TRANSPORT_UART defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /Users/johnboiles/Developer/repos/espressif/esp-idf-54/components/bt/host/nimble/Kconfig.in:977
    /Users/johnboiles/Developer/repos/sindarin-inc/embedded-app/components/esp-nimble-cpp/Kconfig:225
```

They seem to be benign but definitely looked scary.

h2zero, 2025-02-05:
Those are just kconfig warnings because some bluetooth options were added to enable NimBLE support on the esp32p4, not sure if those can be resolved.

johnboiles, 2025-02-05:
Got it, so the warning thing doesn't acknowledge the `if IDF_TARGET_ESP32P4` is true and complains even if those are only relevant to the P4?

h2zero, 2025-02-05:
It would seem that way, I'll try to find a way to avoid the warnings when I get some time.

johnboiles, 2025-02-06:
I'm getting this in other parts of IDF 5.4 too. Feel free to close this if you want!

```
info: INFO: Symbol ESP_WIFI_STATIC_RX_BUFFER_NUM defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /Users/johnboiles/Developer/repos/sindarin-inc/embedded-app/managed_components/espressif__esp_wifi_remote/./idf_v5.4/Kconfig.wifi.in:4
    /Users/johnboiles/Developer/repos/espressif/esp-idf-54/components/esp_wifi/Kconfig:26
info: INFO: Symbol ESP_WIFI_DYNAMIC_RX_BUFFER_NUM defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /Users/johnboiles/Developer/repos/sindarin-inc/embedded-app/managed_components/espressif__esp_wifi_remote/./idf_v5.4/Kconfig.wifi.in:20
    /Users/johnboiles/Developer/repos/espressif/esp-idf-54/components/esp_wifi/Kconfig:42
info: INFO: Symbol ESP_WIFI_STATIC_TX_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /Users/johnboiles/Developer/repos/espressif/esp-idf-54/components/esp_wifi/Kconfig:78
    /Users/johnboiles/Developer/repos/sindarin-inc/embedded-app/managed_components/espressif__esp_wifi_remote/./idf_v5.4/Kconfig.wifi.in:56
info: INFO: Symbol ESP_WIFI_DYNAMIC_TX_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /Users/johnboiles/Developer/repos/espressif/esp-idf-54/components/esp_wifi/Kconfig:80
    /Users/johnboiles/Developer/repos/sindarin-inc/embedded-app/managed_components/espressif__esp_wifi_remote/./idf_v5.4/Kconfig.wifi.in:58
```

h2zero, 2025-02-06:
Those don't look like anything that would be caused by this library, there are no WiFi related definitions here.

johnboiles, 2025-02-06:
Of course, I'm just pointing out there are similar warnings internal to the Espressif libraries, so maybe there's nothing to be done about them.

h2zero, 2025-02-07:
Ahh, okay cool. Would be nice to find a workaround for sure though, they bug me too lol.

wuyuanyi135, 2025-08-14:
I am experiencing the same warnings:
```
warning: default on the choice symbol BT_NIMBLE_ENABLED (defined at C:/Espressif/frameworks/esp-idf-v5.5/components/bt/Kconfig:22, C:/Users/wuyua/CLionProjects/UniLAB/projects/project_unilab_drive/managed_components/h2zero__esp-nimble-cpp/Kconfig:217) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol BT_NIMBLE_ENABLED (defined at C:/Espressif/frameworks/esp-idf-v5.5/components/bt/Kconfig:22, C:/Users/wuyua/CLionProjects/UniLAB/projects/project_unilab_drive/managed_components/h2zero__esp-nimble-cpp/Kconfig:217) is defined with a prompt outside the choice
info: INFO: Symbol BT_ENABLED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    C:\Users\wuyua\CLionProjects\UniLAB\projects\project_unilab_drive\managed_components\h2zero__esp-nimble-cpp\Kconfig:210
    C:\Espressif\frameworks\esp-idf-v5.5\components\bt\Kconfig:3
info: INFO: Symbol BT_NIMBLE_ENABLED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    C:\Users\wuyua\CLionProjects\UniLAB\projects\project_unilab_drive\managed_components\h2zero__esp-nimble-cpp\Kconfig:217
    C:\Espressif\frameworks\esp-idf-v5.5\components\bt\Kconfig:22
info: INFO: Symbol BT_NIMBLE_TRANSPORT_UART defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    C:\Espressif\frameworks\esp-idf-v5.5\components\bt\host\nimble\Kconfig.in:1237
    C:\Users\wuyua\CLionProjects\UniLAB\projects\project_unilab_drive\managed_components\h2zero__esp-nimble-cpp\Kconfig:225
```

Are these new options added in v5.5?? Should we prefer the esp-idf KConfigs?

h2zero, 2025-08-15:
The warnings are benign and do not cause any harm, please ignore them.