h2zero/esp-nimble-cpp #306: Unable to get characteristic despite it being found and having a secure connection

blathers123, 2025-02-13:
Hello,

I'm trying to read a blood glucose level from a Contour next one glucose meter. Unfortunately, I am unable to subscribe to the characteristic even after establishing a secure connection. You can see in the logs that the code is able to find the matching UUID

below are the logs and relevant code snippet. Using release 2.2.1 and esp-idf 5.4.0 on an esp32-wroom-32e

Thanks for your help and hard work!

code:

where Contour_Next_One_BLE is a sub-class of BLESensor

Contour_Next_One_BLE:
```C++
Contour_Next_One_BLE::Contour_Next_One_BLE()
{
  _service_UUID = NimBLEUUID((uint16_t) 0x1808);
  _characteristic_UUID = NimBLEUUID("00002a04-0000-1000-8000-00805f9b34fb");
  _advertised_service_UUID = _service_UUID;
  _make = "Contour";
  _model_name = "Next One";
  _advertised_device_title = "Contour";
  _advertised_name_is_substring = true; // this device's advertised name includes a serial number
  _requires_secure_connection = true;
  _notify_callback = notifyCallback;
  _client_callback = new Contour_Next_One_BLE::ClientCallbackTyped();
  ESP_LOGD(TAG, "Contour_Next_One_BLE Constructor");
}

void Contour_Next_One_BLE::notifyCallback(BLERemoteCharacteristic *pBLERemoteCharacteristic, uint8_t *pData, size_t length, bool isNotify)
{
  ESP_LOGD(TAG, "Contour_Next_One_BLE::notifyCallback");

  // do stuff with the data
}

void Contour_Next_One_BLE::ClientCallbackTyped::onConnect(NimBLEClient *pclient)
{
  ESP_LOGD(TAG, "Contour_Next_One_BLE onConnect");
}
void Contour_Next_One_BLE::ClientCallbackTyped::onDisconnect(NimBLEClient *pclient, int reason)
{
  ESP_LOGD(TAG, "Contour_Next_One_BLE onDisconnect");
}

void Contour_Next_One_BLE::ClientCallbackTyped::onConfirmPasskey(NimBLEConnInfo& connInfo, uint32_t pass_key) {
  ESP_LOGD(TAG,"The passkey YES/NO number: %" PRIu32 "\n", pass_key);
  /** Inject false if passkeys don't match. */
  NimBLEDevice::injectConfirmPasskey(connInfo, true);
}

/** Pairing process complete, we can check the results in connInfo */
void Contour_Next_One_BLE::ClientCallbackTyped::onAuthenticationComplete(NimBLEConnInfo& connInfo) {
  if (!connInfo.isEncrypted()) {
      ESP_LOGD(TAG,"Encrypt connection failed - disconnecting\n");
      /** Find the client with the connection handle provided in connInfo */
      NimBLEDevice::getClientByHandle(connInfo.getConnHandle())->disconnect();
      return;
  }
}
```
BLESensor:

```C++
/**
 * @brief connects to the provided device and optionally subscribes to notifications from it
 *
 * @param address the address device to connect to
 * @param client an instance of BLEClient to connect
 * @return true if connected, false otherwise
[... 259 lines cut ...]
D (31609) NimBLERemoteService: << retrieveCharacteristics()
D (31609) NimBLERemoteService: >> retrieveCharacteristics()
I (31619) NimBLE: GATT procedure initiated: discover characteristics by uuid; 
I (31629) NimBLE: start_handle=9 end_handle=20 uuid=
I (31629) NimBLE: 0x2a04
I (31639) NimBLE: 

D (31729) NimBLERemoteService: Characteristic Discovery >> status: 14 handle: -1
D (31729) NimBLERemoteService: << Characteristic Discovery
D (31729) NimBLERemoteService: << retrieveCharacteristics()
D (31729) NimBLERemoteService: << Characteristic not found
D (31739) BLESensor: unable to get characteristic: 00002a04-0000-1000-8000-00805f9b34fb
E (31749) BLESensor: service not found. Disconnecting
I (31749) NimBLE: GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19

Failed to connect, starting scan
D (31769) NimBLEScan: >> start: duration=10000
I (31769) NimBLE: GAP procedure initiated: discovery; 
I (31779) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1 
I (31779) NimBLE: duration=10000ms
I (31789) NimBLE: 

D (31799) NimBLEClient: >> handleGapEvent 
D (31799) NimBLEClient: disconnect; reason=534, 
D (31799) Contour_Next_One_BLE: Contour_Next_One_BLE onDisconnect```

h2zero, 2025-02-13:
From the logs it looks like the characteristic is not found, can you read write from this on your phone with nRFconnect or other app?

blathers123, 2025-02-13:
I can see the characteristic with the nRFconnect app (though I have had trouble subscribing to the notification)

If I'm reading the logs right, I first ask for the characteristic with UUID  00002a04-0000-1000-8000-00805f9b34fb

```
D (31439) NimBLERemoteService: << Characteristic Discovery
D (31439) NimBLERemoteService: << retrieveCharacteristics()
D (31449) NimBLERemoteService: >> getCharacteristic: uuid: 00002a04-0000-1000-8000-00805f9b34fb
D (31459) NimBLERemoteService: >> retrieveCharacteristics()
```

Then it finds the characteristic I want 

```
I (31459) NimBLE: GATT procedure initiated: discover characteristics by uuid; 
I (31469) NimBLE: start_handle=9 end_handle=20 uuid=
I (31479) NimBLE: 00002a04-0000-1000-8000-00805f9b34fb
```
and it finds another characteristic
```
I (31619) NimBLE: GATT procedure initiated: discover characteristics by uuid; 
I (31629) NimBLE: start_handle=9 end_handle=20 uuid=
I (31629) NimBLE: 0x2a04
I (31639) NimBLE: 
```

Then says `D (31729) NimBLERemoteService: << Characteristic not found`

h2zero, 2025-02-14:
The logs show the following:

>D (31379) NimBLERemoteService: Characteristic Discovery >> status: 0 handle: 10
D (31379) NimBLERemoteService: Characteristic Discovery >> status: 0 handle: 13
D (31389) NimBLERemoteService: Characteristic Discovery >> status: 0 handle: 16
D (31409) NimBLERemoteService: Characteristic Discovery >> status: 0 handle: 18
D (31439) NimBLERemoteService: Characteristic Discovery >> status: 14 handle: -1
D (31439) NimBLERemoteService: << Characteristic Discovery
D (31439) NimBLERemoteService: << retrieveCharacteristics()

All characteristices found for this service, next get the characteristic with the uuid: 

>D (31449) NimBLERemoteService: >> getCharacteristic: uuid: 00002a04-0000-1000-8000-00805f9b34fb
D (31459) NimBLERemoteService: >> retrieveCharacteristics() // uhoh that UUID wasn't found in the original discovery, try again:

>D (31609) NimBLERemoteService: Characteristic Discovery >> status: 14 handle: -1
D (31609) NimBLERemoteService: << Characteristic Discovery
D (31609) NimBLERemoteService: << retrieveCharacteristics()

Not found with 128 bit uuid, trying 16bit:

>D (31609) NimBLERemoteService: >> retrieveCharacteristics()
I (31619) NimBLE: GATT procedure initiated: discover characteristics by uuid; 
I (31629) NimBLE: start_handle=9 end_handle=20 uuid=
I (31629) NimBLE: 0x2a04
I (31639) NimBLE: 
D (31729) NimBLERemoteService: Characteristic Discovery >> status: 14 handle: -1
D (31729) NimBLERemoteService: << Characteristic Discovery
D (31729) NimBLERemoteService: << retrieveCharacteristics()
D (31729) NimBLERemoteService: << Characteristic not found

Still not found, giving up.

So there is something wrong with the UUID you are requesting or some unknown issue, need to be bonded maybe?

blathers123, 2025-02-14:
I was able to verify using a ble app that the UUID does exist and it does match, 

Very possible it needs to be bonded. I put the sensor into pairing mode and I can see in the logs that it successfully establishes a secure connection but still fails to find the UUID. 

Do I need to do something special for bonding it? I've checked through the documentation for this and the Arduino version and didn't see anything relevant

Also thanks a ton for your help 🙏

h2zero, 2025-02-14:
Please add `ESP_LOGD(TAG, "%s", pSvc->toString().c_str());` after `pSvc->getCharacteristics(true);` and post the output here so we can see what's happening.

Bonding can be enabled with `NimBLEDevice::setSecurityAuth(bool bonding, bool mitm, bool sc);` setting the first parameter to true.

blathers123, 2025-02-17:
I added the debugging log as you suggested and embarrassingly it was a UUID  issue. I'll mark this as closed. Thanks a ton for your help!

closed 2025-02-17 by blathers123