h2zero/esp-nimble-cpp #308: Failing to connect with pClient->connect(&device)

goofy2k, 2025-02-22:
I try to establish a connection between two ESP32 devices. I succesfully ran the NimBLE_Server, NimBLE_Client and NimBLE_Async_Client examples and was able to interact with the server with the nRF app. I integrated server code into my full application and can again interact with the nRF app.  

I tried to integrate the  NimBLE_Client into the application on the peer device, but connection was not successful.  I switched to reading the New_user_guide and integrated the full example code for the client in my own code. Had to change some things, such as changing NimBLEUUID serviceUuid("ABCD");  in the example into NimBLEUUID serviceUuid("dead");  (not "DEAD" !) to be compliant with the UUID in the server code. This helped me to get closer to connecting as the server device was now successfully discovered in NimBLEScanResults results.

However: Still no connection.  Of course, this could have to to with something on the server-side, but I started checking the client code with adding a number of logs. It looks that pClient.connect() is never entered as I don't see logs that are inside that function. I add my code derived from the full example in the Client walkthrough and my output. 

Some help would be appreciated! Thanks.

``int configNimBLEClientWalkthrough(NimBLEClient *&pClient) {
	//returns pClient by reference ! to app_main
	static const char *TAG = "configCLIENT_wt";
	//example from Walkthrough
	int rc = 1;  //default set to error
	NimBLEDevice::init("");
	ESP_LOGW(TAG, "NimBLEDevice::getScan()");
	NimBLEScan *pScan = NimBLEDevice::getScan();
	NimBLEScanResults results = pScan->getResults(10 * 1000);
	ESP_LOGW(TAG, "Nr of scan results: %d", results.getCount());

	NimBLEUUID serviceUuid("dead");  //NOTE:  non-numeric  and NO CAPITALS
	ESP_LOGW(TAG, "Looking for UUID = dead");
	for (int i = 0; i < results.getCount(); i++) {
		ESP_LOGW(TAG, "Looking in result nr: %d", i);
		const NimBLEAdvertisedDevice *device = results.getDevice(i);
		if (device != nullptr) {
			printf("Device: %s\n", device->toString().c_str()); //FV

			if (device->isAdvertisingService(serviceUuid)) {
				ESP_LOGW(TAG,"Found device with the required service. Creating a client for it!");
				NimBLEClient *pClient = NimBLEDevice::createClient();

				if (!pClient) { // Make sure the client was created
					ESP_LOGE(TAG, "Client was not created!");
					break;
				}
				//check validity of pClient pointer
				printf("Client: %s\n", pClient->toString().c_str()); // FV

				ESP_LOGW(TAG,"Trying to connect");
				if (pClient->connect(&device)) { // ✅ Pass pointer directly (no &device vs with &device)
					ESP_LOGW(TAG, "Client connected successfully to device");
					NimBLERemoteService *pService =
						pClient->getService(serviceUuid);

					if (pService != nullptr) {
						ESP_LOGW(TAG, "Service found!");
						NimBLERemoteCharacteristic *pCharacteristic =
							pService->getCharacteristic("beef");

						if (pCharacteristic != nullptr) {
							ESP_LOGW(TAG, "Characteristic found!");
							std::string value = pCharacteristic->readValue();
							// print or do whatever you need with the value
						} else {
							ESP_LOGE(TAG, "Characteristic not found!");
						}
					} else {
						ESP_LOGE(TAG, "Service not found!");
					}
				} else {
					// failed to connect
					ESP_LOGE(TAG, "Failed to connect client to device");
				}
				ESP_LOGW(TAG, "Deleting client!");
				NimBLEDevice::deleteClient(pClient);
			}
		} else {
			ESP_LOGE(TAG, "device is nullptr");
			return rc;
		}

	}
	
    return rc; // do usefull checks to assure that 
		//returning 0 is valid here

The output:
  
```
W (1554) app_main: Calling configNimBLEClientWalkthrough
I (1574) BTDM_INIT: BT controller compile version [b022216]
I (1574) BTDM_INIT: Bluetooth MAC: 84:cc:a8:0d:3b:ba
I (1574) phy_init: phy_version 4830,54550f7,Jun 20 2024,14:22:08
W (1924) configCLIENT_wt: NimBLEDevice::getScan()
I (1924) NimBLE: GAP procedure initiated: discovery;
I (1924) NimBLE: own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=1
I (1934) NimBLE: duration=10000ms
I (1944) NimBLE:

W (11954) configCLIENT_wt: Nr of scan results: 29
W (11954) configCLIENT_wt: Looking for UUID = dead
W (11954) configCLIENT_wt: Looking in result nr: 0
Device: Name: , Address: 5d:7b:5a:6b:28:d0, manufacturer data: 7500021861b1f7d804a32b3c3ce1044633f8aa1ab3018117fbb1
W (11964) configCLIENT_wt: Looking in result nr: 1
Device: Name: , Address: 6e:20:f9:41:92:a1
W (11974) configCLIENT_wt: Looking in result nr: 2
Device: Name: midi_sequencer_V2_1, Address: 84:cc:a8:0d:3b:72, serviceUUID: 0xdead        <<<<<<<< this is my server
W (11984) configCLIENT_wt: Found device with the required service. Creating a client for it!
Client: peer address: 00:00:00:00:00:00                                  <<<<<<<<pointer is OK
Services:

W (12004) configCLIENT_wt: Trying to connect
                                                                               <<<<<<<< no logs from connect
E (12004) configCLIENT_wt: Failed to connect client to device
W (12014) configCLIENT_wt: Deleting client!
W (12014) configCLIENT_wt: Looking in result nr: 3
Device: Name: , Address: 36:7e:84:32:66:85, manufacturer data: 0600010f212a41f5b88ae9d4467265645f4c656e6f766f
W (12034) configCLIENT_wt: Looking in result nr: 4
Device: Name: , Address: 03:86:b0:5b:22:83
Service Data:
```

goofy2k, 2025-02-22:
Exactly the same happens in the "almost vanilla" NimBLE_Client example. The  

if (!pClient->connect(advDevice)) does not deliver!

goofy2k, 2025-02-22:
When I do:

pClient->connect(advDevice->getAddress());  in stead of pClient->connect(advDevice); things go better!

static const NimBLEAdvertisedDevice* advDevice;   so advDevice is a pointer at an object of type  NimBLEAdvertisedDevice,  while
NimBLEClient::connect  has a header:

    bool NimBLEClient::connect(const NimBLEAddress& address, bool deleteAttributes, bool asyncConnect, bool exchangeMTU)

There may be more locations in the code where these variables are mixed up!

h2zero, 2025-02-22:
Thanks for reporting this, there is a fix in progress.

goofy2k, 2025-02-22:
A few words on using character strings ("BEEF") for entering UUIDs  instead of literal numerical values (0xbeef) may also help for newbies....  ;-)   .  Furthermore this is an excellent, well structured and well documented lib!

closed 2025-02-22 by goofy2k