h2zero/esp-nimble-cpp #309: WiFi and NimBLE together ?

butaikis, 2025-02-23:
Is it possible to show an example of how to run Wi-Fi and BLE (NimBLE) together? At the moment, after `NimBLEDevice::init("");` get errors
```
[  9065][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0

abort() was called at PC 0x401b17f0 on core 0

Backtrace: 0x400837de:0x3ffcc700 0x401394dd:0x3ffcc720 0x400983ea:0x3ffcc740 0x401b17f0:0x3ffcc7b0 0x4019ee30:0x3ffcc7d0 0x4019c668:0x3ffcc7f0 0x40099d8d:0x3ffcc810
  #0  0x400837de in panic_abort at D:\Develop_VS_Code\.platformio\packages\framework-espidf\components\esp_system/panic.c:408
  #1  0x401394dd in esp_system_abort at D:\Develop_VS_Code\.platformio\packages\framework-espidf\components\esp_system/esp_system.c:137
  #2  0x400983ea in abort at D:\Develop_VS_Code\.platformio\packages\framework-espidf\components\newlib/abort.c:46
  #3  0x401b17f0 in pm_set_sleep_type at ??:?
  #4  0x4019ee30 in wifi_set_ps_process at ??:?
  #5  0x4019c668 in ieee80211_ioctl_process at ??:?
  #6  0x40099d8d in ppTask at ??:?
```

Thanks for any reply!

h2zero, 2025-02-24:
There isn't anything special required for this, as long as you have coexist enabled in the sdkconfig it works without any work on your part.

butaikis, 2025-02-25:
@h2zero I already found out, the problem was `WiFi.setSleep(WIFI_PS_NONE);` disabled it

closed 2025-02-25 by butaikis