h2zero/esp-nimble-cpp #311: Issue with onWrite Handling 0x00 After Migrating to NimBLE 2.2.1

Oblivious0214, 2025-03-01:
In the previous version (1.4.0), the server did not terminate data reception when encountering 0x00. However, after migrating to the latest version (2.2.1), we’ve noticed that the onWrite function seems to have added a new logic that ends data reception upon receiving 0x00.

We would like to request an option to disable this behavior, as in our legacy project, 0x00 has significant meaning, and this new logic causes issues. It would be helpful to have a manual setting to turn off this behavior to ensure compatibility with older projects.

h2zero, 2025-03-01:
@Oblivious0214 This should not have changed at all. Could you share a code snippet that reproduces this issue?

Oblivious0214, 2025-03-03:
> [@Oblivious0214](https://github.com/Oblivious0214) This should not have changed at all. Could you share a code snippet that reproduces this issue?

Got it. I will refer to the latest example code and rewrite it to verify whether the issue is on my end.

closed 2025-03-06 by Oblivious0214

Oblivious0214, 2025-03-06:
It has been confirmed that the issue arose due to inattention during code migration. After rewriting it according to the 2.x example, it performs normally. One problem was that in 2.x, the syntax for adding characteristics to a Service was different. Although the old version could run, it would automatically truncate after encountering 0x00.