h2zero/esp-nimble-cpp #313: Wi-Fi + BLE resulting in BLE disconnections after some amount of time

cmorganBE, 2025-03-04:
esp32s3, esp-idf v5.3, esp-nimble-cpp 63a3301696 (Add missing const in NimblEConnInfo).

I'm not suggesting this is an esp-nimble-cpp issue but I'm also not sure what to do next in terms of how to gather the next piece of information.

What I'm seeing:

- Wi-Fi connected
- Connect with BLE (iOS + lightblue), pair and bond
- Retrieve some characteristics, works fine
- Sit for a minute or two
- BLE disconnected

It's possible to reconnect as soon as the disconnection occurs, the same thing happens after this reconnection, after a few minutes.

Tried to look at 'NimBLE' output via 'log_level NimBLE debug', and I see this when the disconnection occurs:

```
no new log output here until BLE disconnects, then the below happens

D (2006502) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x5 
D (2006502) NimBLE: 

I (2006502) BLE: Client disconnected
I (2006502) NimBLE: GAP procedure initiated: advertise; 
I (2006502) NimBLE: disc_mode=2
I (2006502) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=2 adv_itvl_min=0 adv_itvl_max=0
I (2006502) NimBLE: 

D (2006502) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0006 len=15

D (2006502) NimBLE: 0x06 
D (2006502) NimBLE: 0x20 
D (2006502) NimBLE: 0x0f 
D (2006502) NimBLE: 0x30 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x60 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x00 
D (2006502) NimBLE: 0x07 
D (2006502) NimBLE: 0x02 
D (2006502) NimBLE: 

D (2006502) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1

D (2006502) NimBLE: 0x0a 
D (2006502) NimBLE: 0x20 
D (2006502) NimBLE: 0x01 
D (2006502) NimBLE: 0x01 
D (2006502) NimBLE: 

I (2006512) EESystem: BLE lastState (3) != state (0)
I (2006512) EESystem: BleRestricted
```

Thoughts on how to gather the data to debug?

h2zero, 2025-03-06:
Using nRFConnect you might be able to see something in the logs on the iOS side. Would be good to see the reason code in the NimBLE logs as well.

cmorganBE, 2025-03-06:
@h2zero how do I enable the reason code from the nimble side? I did turn on debugging as you can see above but maybe this isn't everything?

h2zero, 2025-03-12:
@cmorganBE The logs only show the stack debug logs. Please enable the esp-nimble-cpp debug logs as well. Also the `onDisconnect` callback provides the disconnect reason.

cmorganBE, 2025-03-26:
Modified onDisconnect() to print out the reason code.

```
I (77824) BLE: Client disconnected - reason 0x208 (520)
```

Which from https://mynewt.apache.org/latest/network/ble_hs/ble_hs_return_codes.html appears to indicate:

```
0x0208
0x08
BLE_ERR_CONN_SPVN_TMO
Connection Timeout
```

And sure enough I'm messing with connection parameters:

```
// updateConnParams() to shorten the connection interval to
// improve performance
auto minInterval = 6; // 6 * 1.25 = 7.5ms
auto maxInterval = minInterval;
auto latency_packets = 0;
auto timeout = 10; // 10 * 10ms = 100ms
pServer->updateConnParams(connInfo.getConnHandle(),
                           minInterval,
                           maxInterval,
                           latency_packets,
                           timeout);
pServer->setDataLen(connInfo.getConnHandle(), 251);
```

@michaelboeding I'm guessing I'm violating some iOS rules. After dropping the call to updateConnParams() everything seems stable here. No connections dropped for some time.

h2zero, 2025-03-27:
@cmorganBE Thanks for the update, glad you found the cause. Yes, connection parameters are essential but also problematic.

mickeyl, 2025-05-23:
BLE and iOS is a delicate story, in particular with respect to performance and stability. It doesn't help that iOS comes with its own shares of bugs. We discussed the connection parameters in particular in https://github.com/espressif/esp-idf/issues/12789 and https://github.com/espressif/esp-idf/issues/13637. It seems you need to experiment a lot to find a stable/performant set of values. And be sure to test every major (and sometimes, minor) release as they are still doing changes to the stack.