h2zero/esp-nimble-cpp #315: ESP32P4 hosted with Arduino as IDF component fail to build and find esp_bt.h

DovPearX, 2025-03-07:
The problem occurs when we use arduino as an IDF component but on ESP32P4 some functions are disabled due to the use of hosted by SDIO on esp32c6. The arduino library cannot find #include "esp_bt.h". I probably do not know how to make changes to fix it automatically so please help :D

A temporary solution is to comment out the lines #include "esp_bt.h" (line 29) and esp_bt_controller_mem_release(ESP_BT_MODE_BTDM); from esp32-hal.misc.c (line 310). And also comment out lines 28 to 33 from nimconfig.h

It's also possible that I'm doing something wrong, but I've tried changing various things and nothing works.

iranl, 2025-03-09:
See PR #312 and https://github.com/espressif/arduino-esp32/pull/11035

DovPearX, 2025-03-09:
I just noticed it now, I'm closing it then

closed 2025-03-09 by DovPearX