h2zero/esp-nimble-cpp #319: Getting a panic while exchanging mtu during connection process

blathers123, 2025-03-27:
I'm trying to interface with a problematic BLE sensor. Occasionally, during the connection processes when it's exchanging mtu, I'll get a 
`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.`

I'm running the latest v2.2.1

Logs are below:
```
I (13434) NimBLE: GAP procedure initiated: connect; 
I (13444) NimBLE: peer_addr_type=1 peer_addr=
I (13444) NimBLE: f1:e1:70:1f:fd:07
I (13454) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=12 itvl_max=12 latency=0 supervision_timeout=150 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (13464) NimBLE: 

I (15394) NimBLE: GATT procedure initiated: exchange mtu

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e2820  PS      : 0x00060a30  A0      : 0x800ed820  A1      : 0x3fff4050  
A2      : 0x3fff40fc  A3      : 0x3fff454c  A4      : 0x00000004  A5      : 0x3fff06e9  
A6      : 0x3ffdb600  A7      : 0x3fff40fc  A8      : 0x004cff07  A9      : 0x00000001  
A10     : 0x3ffbdab0  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffe35d0  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x004cff2f  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400e281d:0x3fff4050 0x400ed81d:0x3fff40a0 0x400edaa7:0x3fff40c0 0x400ef98b:0x3fff40f0 0x400f7219:0x3fff4150 0x400ea035:0x3fff4180 0x400e9eca:0x3fff41b0 0x400e944d:0x3fff41e0 0x400e945f:0x3fff4200 0x40145f37:0x3fff4220 0x40081822:0x3fff4240 0x400e2d3b:0x3fff4260 0x40090aca:0x3fff4280
  #0  0x400e281d in NimBLEClient::handleGapEvent(ble_gap_event*, void*) at .pio/libdeps/adafruit_feather_esp32_dev/esp-nimble-cpp/src/NimBLEClient.cpp:1163
  #1  0x400ed81d in ble_gap_call_event_cb at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:896
  #2  0x400edaa7 in ble_gap_call_conn_event_cb at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:922
  #3  0x400ef98b in ble_gap_mtu_event at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:7838
  #4  0x400f7219 in ble_att_svr_rx_mtu at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:822
  #5  0x400ea035 in ble_att_rx at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:517
  #6  0x400e9eca in ble_hs_hci_evt_acl_process at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:1251
  #7  0x400e944d in ble_hs_process_rx_data_queue at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:248
  #8  0x400e945f in ble_hs_event_rx_data at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:545
  #9  0x40145f37 in npl_freertos_event_run at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:451
  #10 0x40081822 in ble_npl_event_run at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
      (inlined by) nimble_port_run at /home/user/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:309
  #11 0x400e2d3b in NimBLEDevice::host_task(void*) at .pio/libdeps/adafruit_feather_esp32_dev/esp-nimble-cpp/src/NimBLEDevice.cpp:835
  #12 0x40090aca in vPortTaskWrapper at /home/user/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

blathers123, 2025-03-27:
It's possible I've solved this by connecting with exchangeMTU=false, and calling exhangeMTU() manually after there is a successful connection.

h2zero, 2025-03-27:
Do you perhaps delete the client and or the callbacks somewhere in your code? This very much reminds me of this: https://github.com/h2zero/NimBLE-Arduino/issues/900