h2zero/esp-nimble-cpp #320: Disconnect callback not always called when building with latest on esp-idf `release/v5.4`

finger563, 2025-03-27:
We're investigating updating from `v5.4.0` tag in esp-idf to `release/v5.4` because it brings in a lot of improvements and bugfixes to the nimble stack (including some nvs corruption issues we'd seen).

However, we've found that with those updates, the `onDisconnect` callback is not always called where previously it was. This is not the case for all BLE hosts, but it is consistent for a single host. The use case here is that our peripheral device programmatically disconnects from the remote, but we still expect the disconnect callback to be called.

When the remote is a phone for instance, this still works as expected.

When the remote is a BLE dongle (connected to Win10 computer), the disconnect callback is never called, whereas previously it was.

I'm just starting to debug to determine if it's an issue with this library (due to changes upstream) or if it's purely due to changes upstream, but I figured I'd open an issue here to make others aware and possibly get some support if others have seen this or have any other information.

For reference, the specific commit hash on the `release/v5.4` branch we're testing is: `5cbd2a3877f3fe521cf34fbcda2522dd925bd7e5`

I'm testing on an ESP32s3.

h2zero, 2025-03-27:
Thanks @finger563, this seems to be a rampant issue that is causing a lot of problems for many folks. So far as i can determine, the cause is in the controller that is closed source and not generating the event, which is passed to the host (NimBLE) then the library. If you enable debug logs you should see whether the event is generated or not and if not then it's an espressif problem.

finger563, 2025-03-28:
Yea, based on the debug logs it seems that the controller doesn't trigger the event:

Failing:
```console
[Hid Service Example/I][12.122]: Button pressed!
I (12344) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

D (12345) NimBLE: ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3

D (12345) NimBLE: 0x06
D (12345) NimBLE: 0x04
D (12346) NimBLE: 0x03
D (12346) NimBLE: 0x01
D (12346) NimBLE: 0x00
D (12346) NimBLE: 0x13
D (12346) NimBLE:

D (12357) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x5
D (12357) NimBLE:
```

[image]

Succeeding:
```console
[Hid Service Example/I][16.589]: Button pressed!
I (16810) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

D (16811) NimBLE: ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3

D (16811) NimBLE: 0x06
D (16812) NimBLE: 0x04
D (16812) NimBLE: 0x03
D (16812) NimBLE: 0x01
D (16812) NimBLE: 0x00
D (16812) NimBLE: 0x13
D (16812) NimBLE:

D (16830) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x5
D (16830) NimBLE:

D (16831) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (16831) NimBLEServer: subscribe event; attr_handle=8, subscribed: false
D (16831) NimBLEServer: << handleGapEvent
D (16832) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (16832) NimBLEServer: subscribe event; attr_handle=24, subscribed: false
D (16832) NimBLECharacteristicCallbacks: onSubscribe: default
D (16832) NimBLEServer: << handleGapEvent
D (16832) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (16833) NimBLEServer: subscribe event; attr_handle=28, subscribed: false
D (16833) NimBLECharacteristicCallbacks: onSubscribe: default
D (16833) NimBLEServer: << handleGapEvent
D (16833) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_SUBSCRIBE
I (16834) NimBLEServer: subscribe event; attr_handle=54, subscribed: false
D (16834) NimBLECharacteristicCallbacks: onSubscribe: default
D (16834) NimBLEServer: << handleGapEvent
D (16834) NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_DISCONNECT
[Hid Service Example/I][16.614]: Device disconnected: LOCAL_USER_TERMINATED
D (16836) NimBLEAdvertising: >> Advertising start: duration=0, dirAddr=NULL
I (16836) NimBLE: GAP procedure initiated: advertise;
I (16836) NimBLE: disc_mode=2
I (16837) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (16837) NimBLE:
```

[image]

finger563, 2025-03-28:
https://github.com/espressif/esp-idf/issues/15670

finger563, 2025-04-01:
There is a patch on that PR which fixes the issue for me - it's a patch for the `nimble` submodule of `esp-idf`.