h2zero/esp-nimble-cpp #321: Failure to pair to Iphone

finger563, 2025-04-08:
Whereas previously my [hid_service/example](https://github.com/esp-cpp/espp/tree/main/components/hid_service/example) would pair and work with my iPhone over BLE every time, now it only works _every other time_. 

Flash the example, then run the following steps to reproduce:
1. Pair to iphone - should see that it bonds and connects just fine
2. Forget pairing info on iphone
3. Attempt to re-pair to iphone - should see that it gives pairing prompt then briefly shows connected before immediately disconnects (`REMOTE_USER_TERMINATED`).
4. Step 3 will continue to fail until you reboot the esp32, at which point it will succeed again, but only once. Any kind of reboot seems to work (power cycle, press reset button, software reset, deep sleep reset)

Logs:

[success.log](https://github.com/user-attachments/files/19651801/success.log)
[failure.log](https://github.com/user-attachments/files/19651800/failure.log)

finger563, 2025-04-08:
And I have confirmed that this behavior persists even after I run `idf.py erase-flash`

finger563, 2025-04-08:
I've opened an upstream issue [here](https://github.com/espressif/esp-idf/issues/15740) as I try to determine if it's an issue in the latest version of `NimBLE` or if it's a required change in `esp-nimble-cpp`

finger563, 2025-04-08:
Differences I've found:

## LINK_ESTAB
### Failure
```console
D (25222) NimBLE: host tx hci data; handle=1 length=8

D (25222) NimBLE: ble_hs_hci_acl_tx():
D (25222) NimBLE: 0x01
D (25222) NimBLE: 0x00
D (25222) NimBLE: 0x08
D (25222) NimBLE: 0x00
D (25222) NimBLE: 0x04
D (25223) NimBLE: 0x00
D (25223) NimBLE: 0x04
D (25223) NimBLE: 0x00
D (25223) NimBLE: 0x1b
D (25223) NimBLE: 0x1c
D (25223) NimBLE: 0x00
D (25223) NimBLE: 0x05 <--
D (25223) NimBLE:
```
### Success
```console
D (6502) NimBLE: host tx hci data; handle=1 length=8

D (6502) NimBLE: ble_hs_hci_acl_tx():
D (6502) NimBLE: 0x01
D (6502) NimBLE: 0x00
D (6502) NimBLE: 0x08
D (6503) NimBLE: 0x00
D (6503) NimBLE: 0x04
D (6503) NimBLE: 0x00
D (6503) NimBLE: 0x04
D (6503) NimBLE: 0x00
D (6503) NimBLE: 0x1b
D (6503) NimBLE: 0x1c
D (6503) NimBLE: 0x00
D (6503) NimBLE: 0x07 <--
D (6503) NimBLE:
```

finger563, 2025-04-08:
Note: this does not happen on esp-idf `v5.4`, but does happen on `v5.4.1`

finger563, 2025-04-08:
Have confirmed this is an esp-idf / nimble upstream issue since this behavior is exhibited in the `esp-idf/examples/bluetooth/nimble/bleprph` example.

https://github.com/espressif/esp-idf/issues/15740#issuecomment-2786946928

h2zero, 2025-04-08:
Thanks @finger563 looks like there are a lot of breaking changes in 5.4.1, lets keep this open for tracking.

finger563, 2025-04-08:
FWIW i'm tracking that the DHKEY check is failing, causing the stack to stall out / go into a bad state preventing the bonding from succeeding. I'll continue to update as I find out more, but the majority of my testing / findings is on the linked issue.

h2zero, 2025-04-08:
What should be happening is that with the phone bond deleted the stack should recognize this and trigger a repairing event that will delete the local bond info and start the process anew. I wonder what they changed.

finger563, 2025-04-09:
For anyone else experiencing this issue, there is a nimble patch available which (in my testing) fixes the issue

https://github.com/espressif/esp-idf/issues/15740#issuecomment-2789983990

finger563, 2025-04-13:
Note: there's a better patch available as that patch breaks pairing compatibility with macos

whni, 2025-04-13:
> Note: there's a better patch available as that patch breaks pairing compatibility with macos

Hi thank you so much for raising this issue. I am also experiencing this issue on macos. What will the better patch will? Thank you!

finger563, 2025-04-13:
> > Note: there's a better patch available as that patch breaks pairing compatibility with macos
> 
> Hi thank you so much for raising this issue. I am also experiencing this issue on macos. What will the better patch will? Thank you!

```diff
diff --git a/nimble/host/src/ble_hs_conn.c b/nimble/host/src/ble_hs_conn.c
index ff4183483..eb749c41a 100644
--- a/nimble/host/src/ble_hs_conn.c
+++ b/nimble/host/src/ble_hs_conn.c
@@ -480,13 +480,11 @@ ble_hs_conn_addrs(const struct ble_hs_conn *conn,
     case BLE_ADDR_PUBLIC_ID:
         addrs->peer_id_addr.type = BLE_ADDR_PUBLIC;
         addrs->peer_ota_addr = conn->bhc_peer_rpa_addr;
-        addrs->peer_ota_addr.type = conn->bhc_peer_addr.type;
         break;

     case BLE_ADDR_RANDOM_ID:
         addrs->peer_id_addr.type = BLE_ADDR_RANDOM;
         addrs->peer_ota_addr = conn->bhc_peer_rpa_addr;
-        addrs->peer_ota_addr.type = conn->bhc_peer_addr.type;
         break;

     default:
```