h2zero/esp-nimble-cpp #332: onRead() Callback Fails After setValue() with Data Exceeding MTU

x-radio, 2025-05-07:
Hi folks,

I've stumbled upon an interesting issue with NimBLE. When calling setValue() with data larger than 21 bytes (e.g., setValue("abcdefghijklmnopqrstu")), the onRead() callback stops being triggered. This happens before any MTU change request, when the MTU is still at its default of 23. The characteristic remains readable (e.g., via nRF Connect), but the onRead() callback is no longer called.

Interestingly, if I increase the MTU via a request, everything starts working again. I suspect the issue lies in the append() method, specifically when it sets m_attr_len to a value greater than MTU-2. I’ve confirmed that commenting out the append() call in setValue() prevents the issue entirely.

Here’s a quick repro:

Call setValue("abcdefghijklmnopqrstu") (22 bytes) with default MTU=23.
Try reading the characteristic — the value is returned, but onRead() isn’t triggered.
Increase MTU (e.g., to 100), and onRead() starts working again.
I think adding an MTU check in append() to reject values larger than MTU-2 could fix this.

Thanks!

h2zero, 2025-05-07:
Thanks @x-radio, is this using the master branch or the release?

h2zero, 2025-05-08:
Okay, thanks. This should be fixed in the master branch for next release.