h2zero/esp-nimble-cpp #333: compiling issue after updating nimble c++ to the latest status

aygh4266, 2025-05-08:
Hi,

I have mentioned before in #322 that getValue() doesn't work correctly with std::string. After updating NimBLE C++ to the latest status I have the following issue

```cpp
C:/Project/esp-nimble-cpp/src/NimBLEAttValue.h: In instantiation of 'T NimBLEAttValue::getValue(time_t*, bool) const [with T = unsigned char; time_t = long long int]':
C:/Project/esp-nimble-cpp/src/NimBLELocalValueAttribute.h:113:35:   required from 'T NimBLELocalValueAttribute::getValue(time_t*, bool) const [with T = unsigned char; time_t = long long int]'
  113 |         return m_value.getValue<T>(timestamp, skipSizeCheck);
      |                ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Project/CharacteristicsCallbacks.cpp:574:57:   required from here
  574 |         eventsrequestValue = pCharacteristic->getValue<uint8_t>();
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
C:/Project/esp-nimble-cpp/src/NimBLEAttValue.h:317:21: error: invalid user-defined conversion from 'const NimBLEAttValue' to 'unsigned char' [-fpermissive]
  317 |             return *this;
      |                     ^~~~
C:/Project/esp-nimble-cpp/src/NimBLEAttValue.h:338:5: note: candidate is: 'NimBLEAttValue::operator const uint8_t*() const' (near match)
  338 |     operator const uint8_t*() const { return m_attr_value; }
      |     ^~~~~~~~
C:/Project/esp-nimble-cpp/src/NimBLEAttValue.h:338:5: note:   no known conversion from 'const uint8_t*' {aka 'const unsigned char*'} to 'unsigned char'
[829/952] Building CXX object esp-idf/bluetooth/CMakeFiles/__idf_bluetooth.dir/services.cpp.obj
ninja: build stopped: subcommand failed.
```

Thanks

h2zero, 2025-05-08:
Can you please show your code that invokes this?

aygh4266, 2025-05-08:
> Can you please show your code that invokes this?

```cpp
typedef enum _TargetController{
   ESP32,
   ARDUINO,
   STM32,
   TOSHIBA
} TargetController;

class TargetControllerUpdaterCallback : public NimBLECharacteristicCallbacks {
private:
	TargetController* pTargetController;
public:
        void onWrite(NimBLECharacteristic* pCharacteristic, NimBLEConnInfo& connInfo);
};

void TargetControllerUpdaterCallback::onWrite(NimBLECharacteristic* pCharacteristic, NimBLEConnInfo& connInfo)
{
	TargetController targetController = *(this->pTargetController);
	targetController = pCharacteristic->getValue<TargetController>();
}
```

This invokes the compiling error.

PR #334 (2025-05-19): [Bugfix] Attribute getValue fails with some data types
Fixes #333
closed 2025-05-19 by h2zero