h2zero/esp-nimble-cpp #345: Latest Not work

BHLabsLtd, 2025-06-21:
Hi,  

i've been using nimble 2.2.1 for a while and it's been working fine (apart from security auth pin).   when updating to 2.3.0/1 i get a panic abort whenever i try to init BLE.  

This happens in a new project and just NimBLEDevice init in main.
 
i'm using ESP32s3
in Visual Code on M4 Mac Book Pro

platform = espressif32@6.10.0
board = esp32-s3-devkitc-1
PlatformIO Core, version 6.1.18

h2zero/esp-nimble-cpp@2.2.1 Works fine

  
0x00000bac: ?? ??:0
0x4037aa2a: panic_abort at /Users/martyn/.platformio/packages/framework-espidf/components/esp_system/panic.c:408
0x4038bda1: esp_system_abort at /Users/martyn/.platformio/packages/framework-espidf/components/esp_system/esp_system.c:137
0x4039ae95: __assert_func at /Users/martyn/.platformio/packages/framework-espidf/components/newlib/assert.c:47
0x420acc1e: os_msys_init_once at /Users/martyn/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/nimble/src/os_msys_init.c:56 (discriminator 1)
0x420acc7a: os_msys_init at /Users/martyn/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/nimble/src/os_msys_init.c:104
0x420a4925: nimble_port_init at /Users/martyn/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:48
0x42078380: NimBLEDevice::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /Users/martyn/Library/CloudStorage/OneDrive-Personal/BHLabs/Developer/iOpt/Source/Evo/iOpt-Evo-Firmware/.pio/libdeps/debug-nbiot-evo/esp-nimble-cpp/src/NimBLEDevice.cpp:965

```

Example Code: 

    if (!NimBLEDevice::isInitialized()) {
        if (NimBLEDevice::init("Evo") == false) {  // Panic here!
            log_error("Failed to initialise BLE");
            return;
        }
    }

```

h2zero, 2025-06-22:
Hello @BHLabsLtd, I just tested with:
```
platform = espressif32@6.10.0
framework = espidf
board = esp32-s3-devkitc-1
lib_deps = h2zero/esp-nimble-cpp@^2.3.1
```
```
extern "C" void app_main() {
    NimBLEDevice::init("test");
    printf("NimBLE Device initialized");
}
```
I was not able to reproduce the issue...

BHLabsLtd, 2025-06-23:
Hmm wonder why mine crashes every time.  I’ll try a clean install of platform io & visual code with just nimble installed.

closed 2025-06-23 by BHLabsLtd