h2zero/esp-nimble-cpp #348: Standard NimBLE C services?

malachib, 2025-07-09:
Hi, and thank you for this library.

We are using a rather old version of `esp-nimble-cpp` for a custom service along with ESP-IDF v5.4.2.

I wanted to use some off-the-shelf NimBLE services, as provided by:

* `ble_svc_dis_init`
* `ble_svc_bas_init`

However, doing so appears to interfere with `esp-nimble-cpp` operations and GATT characteristic reads die once DIS is interacted with.  Is it possible for these to co-exist?  Any idea what's wrong?

h2zero, 2025-07-09:
This could work but you would have to register callbacks for handling lots of things and probably in a correct order, not really recommended. 

Instead I can point you to this: https://github.com/h2zero/NimBLEOta which contains a DIS service that you can copy/modify. The BAS service could be implemented similarly.

malachib, 2025-07-10:
Not the answer I hoped for, but the answer I needed.  Thank you for this