h2zero/esp-nimble-cpp #351: NimBLEScan returning duplicates, even though it's configured not to

blathers123, 2025-07-23:
I've setup the scan like so:
```
// configure the scan
    _scanner = NimBLEDevice::getScan();

    _scanner->setMaxResults(0xFF);

    /** Set the callbacks to call when scan events occur, no duplicates */
    _scanner->setScanCallbacks(&_scan_callbacks, false);

    /** Set scan interval (how often) and window (how long) in milliseconds */
    _scanner->setInterval(100);
    _scanner->setWindow(100);

    // ignore already seen devices
    _scanner->setDuplicateFilter(1);

    /**
     * Active scan will gather scan response data from advertisers
     *  but will use more energy from both devices
     */
    _scanner->setActiveScan(true);
```
I'm still getting onResult() called for BLE Devices that I've already seen.

blathers123, 2025-07-25:
The devices are repeated pretty much immediately. The "Maximum number of devices in scan duplicate filter" is 100 in the espidf menu config. I'm don't think this is due to the controller being full. 
I'm using an esp32-s3 with IDF 5.4.1

h2zero, 2025-07-26:
Can you post a debug level log that shows this?

h2zero, 2025-07-28:
I can see how this is happening, you are restarting the scan, which clears the duplicate filter.

blathers123, 2025-07-28:
I'm using this line to continue the scan:
`_scanner->start(SCAN_DURATION, true, false)`
isContinue is set to true, so it shouldn't clear the duplicate filter, right?

h2zero, 2025-07-29:
The issue is what I described above. When you stop and then restart the scan it clears the duplicate cache and the same device will be discovered again each time you do this. You should analyze the device in the callback or some other means and not stop the scan until ready to connect or something else.

blathers123, 2025-07-29:
Gotcha, I switched that around as you recommended and it's working great. Thanks for your help!

closed 2025-07-29 by blathers123