h2zero/esp-nimble-cpp #353: `CONFIG_BT_NIMBLE_ROLE_CENTRAL` is required with esp-idf 5.5

larryk3, 2025-08-07:
I updated to the latest version of esp-idf (`v5.5`) and the latest esp-nimble-cpp (`2.3.1`), and I needed to to enable `CONFIG_BT_NIMBLE_ROLE_CENTRAL=y` to make my code compile.

The latest release of esp-idf (`v5.5`), updated the underlying nimble library. In particular, it added `#if MYNEWT_VAL(BLE_GATTC)` around some of the deprecated functions:

https://github.com/espressif/esp-nimble/blob/cc3ac5416b46e6e9363dae357378c03b4975cfdb/nimble/host/src/ble_gattc.c#L5446

`esp-nimble-cpp` calls the deprecated `ble_gattc_...` instead of `ble_gatts_...` function. The `ble_gattc_...` function calls the `ble_gatts_...` directly, so it seems like `esp-nimble-cpp` could just call the `ble_gatts_...` function directly.

I believe the only place the `ble_gattc_indicate_custom` and `ble_gattc_notify_custom` are called are in this file:

https://github.com/h2zero/esp-nimble-cpp/blob/master/src/NimBLECharacteristic.cpp#L281

h2zero, 2025-08-10:
Thanks @larryk3, I will update this accordingly

iranl, 2025-08-12:
I stand corrected on this function.

But the `MYNEWT_VAL(....)` macros where added in the mentioned commit and are the reason you need to enable both peripheral and central to get esp-nimble-cpp to compile starting with esp-idf 5.5. And these are the prime reason you are having issues, not the fact that this function has been deprecated for a long time.

larryk3, 2025-08-12:
> And these are the prime reason you are having issues, not the fact that this function has been deprecated for a long time.

Sorry if my original post was not clear. I meant to say that the deprecated function are now surrounded by the macros and switching to the new function will avoid the macro.

PR #355 (2025-08-25): Fix build with idf v5.5+ and specific roles are defined.
Fixes #353