h2zero/esp-nimble-cpp #360: Disconnected client not found with peer RPA enabled

gkoh, 2025-09-06:
esp-nimble-cpp: 2.3.1

In my case when the connection peer has RPA enabled, the GAP disconnect event was not finding the right NimBLEClient and would log:
```
[13:43:33]E NimBLEClient: Disconnected client not found, conn_handle=0
```
when I expected it to call the `onDisconnect()` callback.

After poking around, I reviewed the handling of `BLE_GAP_EVENT_DISCONNECT` in `NimBLEClient::handleGapEvent()`:
```
       case BLE_GAP_EVENT_DISCONNECT: {

            // workaround for bug in NimBLE stack where disconnect event argument is not passed correctly
            pClient = NimBLEDevice::getClientByPeerAddress(event->disconnect.conn.peer_ota_addr);
            if (pClient == nullptr) {
                pClient = NimBLEDevice::getClientByPeerAddress(event->disconnect.conn.peer_id_addr);
            }
            
            if (pClient == nullptr) {
                NIMBLE_LOGE(LOG_TAG, "Disconnected client not found, conn_handle=%d",
                            event->disconnect.conn.conn_handle);
                return 0;
            }   
```

After adding some debug, `peer_ota_addr` and `peer_id_addr` were both set to the identity resolved address, however the expected matching entry in the client list had a peer address of the original resolvable public address, so it would never match.

I fixed my use case by patching to search by handle instead:
```
pClient = NimBLEDevice::getClientByHandle(event->disconnect.conn.conn_handle);
```

I'm not sure if this is the 'right' fix.

I also considered using `NimBLEClient::setPeerAddress()` to update the peer address during `BLE_GAP_EVENT_IDENTITY_RESOLVED`.

gkoh, 2025-09-07:
> Thanks [@gkoh](https://github.com/gkoh), I think both of those options should be done.

@h2zero I can submit a PR for this.
In the disconnect event would you prefer retaining the existing address comparisons or just replace them with the `getByHandle()`?

h2zero, 2025-09-08:
Might as well change them all to getByHandle, which is more efficient .