h2zero/esp-nimble-cpp #364: Only able to establish a single bond

chmorgan, 2025-09-09:
esp32-s3, esp-nimble-cpp v2.3.3

Tested with esp-idf:

* v5.4
* v5.4.1
* v5.4.2
* v5.5.1

All behave the same.

# Steps to reproduce

1. Activate pairing mode on the device (pAdvertising->setScanFilter((false, false)); // no whitelist
1. Connect, pair and bond iPhone via lightblue (app code adds address to whitelist)
1. Can see the bond and whitelist entries show up via NimBLEDevice::getBondedAddress() and getWhiteListAddress() iterated over for getNumBonds() and getWhiteListCount().
1. Able to retrieve characteristics

1. Disconnect from device
1. Re-active pairing mode
1. Connect with PC
1. Connect, pair and bond
1. Now when looking at the bonds there is only 1 bond (there should be two)
1. PC can connect
1. Mobile device cannot

Order can be reversed but the outcome is the same.

# Console output from debug command that I use to introspect the bond and whitelist info

```
bond count: 1
	0 - 14:1b:a0:1c:0f:69
white list entries: 2
	0 - 20:23:51:d6:c4:74
	1 - 14:1b:a0:1c:0f:69
```

Am I missing something relative to telling esp-nimble-cpp to save more than 1 bond?

```
CONFIG_BT_NIMBLE_WHITELIST_SIZE=12
CONFIG_BT_NIMBLE_MAX_BONDS=10
```

h2zero, 2025-09-10:
Interesting, have you tried doing a full flash erase?

Something must have changed upstream if that doesn't fix it.

chmorgan, 2025-09-10:
I haven't tried a full flash erase yet. Let me try that and attempt to reproduce again. It's really easy to reproduce though so that's a bonus. If there is still an issue might want to hire you to fix it if you are interested again @h2zero

chmorgan, 2025-09-14:
@h2zero alright, so I've resurrected an example here and the bond count continues to increase as I pair devices and it looks like its functioning properly.
 
1. Reviewed production code and it's the same in terms of auth and setup.
2. I added several characteristics to the test program and code to adjust their values every second to make it closer to how the production coefficient works.
3. Added wifi and wifi event handling to test program.

Test case code works fine on the test board. Production code does not. I can switch back and forth and the issue shifts back and forth with the code being loaded.

I also copied over a handful of the BT / NIMBLE sdkconfig.defaults.

What kind of debugging can I enable to get more visibility into what's different in terms of behavior or usage?

Possible to get an hour of your time, paid of course, to review and debug, via google meet?