h2zero/esp-nimble-cpp #366: Device without a NimBLEAdvertising::setName() remains unnamed until connecting

chmorgan, 2025-09-13:
esp-idf v5.4.2, esp-nimble-cpp v2.3.3

Unlike the example I wasn't calling

```
    NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
    pAdvertising->setName("testingabcd");
```

but I was calling 

```
NimBLEDevice::init("some name here");
```

Without the call to NimBLEAdvertising::setName() the device shows up as Unnamed but when I connect the name appears.

I don't think I spotted it in years of using the library because I tend to test on the same device all of the time and I think iPhone caches the device name, but it does explain a case where I changed the naming convention on the device, changing the value passed to init() slightly, and the name only changed AFTER connecting to the device.

Is this intentional? I might suggest that internally if NimBLEAdvertising::setName() isn't called, that the NimbleDevice deviceName be used instead.

h2zero, 2025-09-13:
The name  in the call to 'init` sets the device name characteristic value that connecting peers will read from and usually takes precedence over the name in the advertisement when viewing from a phone app etc, which can be different.

chmorgan, 2025-09-14:
Ahh. So it's up to the end device that will decide between the device name characteristic vs. the advertising name?

My mind always goes to what I'd put in the documentation to help the next person. Would something like this make sense:

"
The device will appear with a name based on your particular platform. For example, iOS will show the name set via NimBLEAdvertising::setName() when scanning, but will switch to the device name set via NimBLEDevice::init() after connected.

Note: If you only set the device name via NimBLEDevice::init() your device may show up as 'Unnamed' until connected.
"

h2zero, 2025-09-14:
That is a very accurate description of the situation for sure.

closed 2025-09-19 by chmorgan