h2zero/esp-nimble-cpp #369: Error while deleting whitelist entries, rc=524

chmorgan, 2025-09-16:
I print out the whitelist and bond entries, then run 'ble_bond_delete' to delete the bonds and am getting a rc=524, and a BLE_ERR_CMD_DISALLOWED error.

Is the deletion logic broken somehow? I always delete the first entry until there are zero whitelist entries, as there is no way to delete by position, which makes sense as whitelist entries don't really have an order.

```
static int ble_bond_delete(int argc, char **argv)
{
    printf("Deleting bonds\n");

    auto ret = NimBLEDevice::deleteAllBonds();
    int retval = -1;
    if(ret)
    {
        printf("Bonds deleted\n");
        retval = 0;
    } else
    {
        printf("Error reported while deleting bonds\n");
    }

    printf("Clearing whitelist\n");

    if(NimBLEDevice::getWhiteListCount() == 0)
    {
        printf("No whitelist entries to delete\n");
    }

    while(NimBLEDevice::getWhiteListCount())
    {
        auto address = NimBLEDevice::getWhiteListAddress(0);
        ret = NimBLEDevice::whiteListRemove(address);
        if(ret)
        {
            printf("Deleted whiltelist entry\n");
        } else
        {
            printf("Error deleting whitelist entry\n");
            retval = -1;
            break;
        }
    }

    return retval;
}
```

```
esp> ble_info
Device name: 'SYM-EMV-4C74'
ble state: 'Restricted'
Connect mode: 'WhitelistOnly'
Total changes: 54, saved_notifies: 22
bond count: 2
	0 - 70:ea:5a:15:6a:3f
	1 - 14:1b:a0:1c:0f:69
white list entries: 2
	0 - 70:ea:5a:15:6a:3f
	1 - 14:1b:a0:1c:0f:69
esp> ble_dW (13100) BLE: 2
esp> ble_bond_delete
Deleting bonds
I (15590) LoRaWAN: Modem event callback
W (15590) LoRaWAN: Event received: JOINFAIL
Bonds deleted
Clearing whitelist
I (15970) NimBLE: GAP procedure initiated: set whitelist; 
I (15970) NimBLE: count=1 
I (15970) NimBLE: entry-0={addr_type=0 addr=
I (15970) NimBLE: 14:1b:a0:1c:0f:69
I (15970) NimBLE: } 
I (15970) NimBLE: 

I (15970) NimBLE: ogf=0x08, ocf=0x0010, hci_err=0x20C : BLE_ERR_CMD_DISALLOWED (Command Disallowed)

E (15970) NimBLEDevice: Failed removing from whitelist rc=524
Error deleting whitelist entry
Command returned non-zero error code: 0xffffffff (ESP_FAIL)
```

h2zero, 2025-09-16:
I believe this a response from the controller, I can't find anything in the host code that would return this error from the related functions. Are you already advertising or something when you call this?

chmorgan, 2025-09-16:
@h2zero yep, advertising at this point but nothing connected. Does advertising have to be stopped when performing the bond / whitelist deletion? If so I can shift it into the ble management class and out of the cli scope so I can stop, delete, and start advertising.

h2zero, 2025-09-16:
You cannot be advertising when you change the whitelist, you can remove bonds though.

chmorgan, 2025-09-16:
ahh gotcha. Any thought on if the library should provide better info in that case or catch it and return a proper error code before calling the function? It may not be apparent to clueless devs like myself that this is constraint.

h2zero, 2025-09-17:
Best I think that could be done is an error message. If scanning or advertising is active then the whitelist cannot be manipulated.

closed 2025-09-19 by chmorgan